open-power/op-test #7: Build verification tests fails on Open Power Firestone Hardware

PavamanSubramaniyam, 2015-11-05:
I am executing the BVT tests targeted for a Open Power Firestone Hardware as below.

[root@lop827 bvt]# ./run-op-bvt --bmcip 9.40.193.47 --bmcuser sysadmin --bmcpwd superuser --usernameipmi ADMIN --passwordipmi admin --cfgfiledir ../ci/source/ --imagedir /tmp/ --imagename firestone_update.pnor --fverbose /tmp/new_firestone ./op-ci-basic-bvt.xml

ipmitool -H 9.40.193.47 -I lanplus -U ADMIN -P admin sdr elist |grep 'OCC Active'
ipmitool -H 9.40.193.47 -I lanplus -U ADMIN -P admin sdr elist |grep 'OCC Active'
ipmitool -H 9.40.193.47 -I lanplus -U ADMIN -P admin sdr elist |grep 'OCC Active'
ipmitool -H 9.40.193.47 -I lanplus -U ADMIN -P admin sdr elist |grep 'OCC Active'
IPL timeout
00:26:29 :: op-ci-basic: ERROR: command failed with exit code 256
00:26:29 :: op-ci-basic: Finished Test:
00:26:29 :: op-ci-basic: Starting Test:
00:26:29 :: op-ci-basic: Running command: op-ci-bmc-run  "op_ci_bmc.ipmi_sel_check()"
ipmitool -H 9.40.193.47 -I lanplus -U ADMIN -P admin sel elist
00:26:45 :: op-ci-basic: OK: command finished successfully
00:26:45 :: op-ci-basic: Finished Test:
00:26:45 :: op-ci-basic: ERROR: Finished tests on 2015-November-04 with one or more testcase errors.
00:26:45 :: ERROR: "OP CI Basic BVT" (op-ci-basic.xml) finished with one or more errors
00:26:45 :: ERROR: Finished BVT tests on 2015-November-04 with 1 testcase failures.
All logs and FFDC saved under /tmp/root/bvt/2015-11-04-89279

As can be seen the tests fail because of the failure of the command:
ipmitool -H 9.40.193.47 -I lanplus -U ADMIN -P admin sdr elist |grep 'OCC Active'

As per my analysis, I have found out that the above command is failing because the 
grep 'OCC Active' is failing to fetch the correct output from the sdr elist of the Firestone Hardware.

However, on a Habanero Hardware the output looks as follows:
[root@lop826 src]# ./ipmitool -H 9.47.71.252 -I lanplus -U ADMIN -P admin sdr elist |grep 'OCC Active'
OCC Active       | 08h | ok  | 210.0 | Device Enabled

There are 2 OCC chips on Firestone and 1 OCC chip on Habanero.
In the sdr elist entries there will be OCC 1 Active and OCC 2 Active mentioned for Firestones.

So, the correct fix for the above command to work for both Habanero and Firestone Hardware's are:
=======================================================================
[root@lop826 src]# ./ipmitool -H 9.40.193.53 -I lanplus -U ADMIN -P admin sdr elist |grep '^OCC'
OCC 1 Active     | 08h | ok  | 210.0 | Device Enabled
OCC 2 Active     | 09h | ok  | 210.1 | Device Enabled
[root@lop826 src]# ./ipmitool -H 9.47.71.252 -I lanplus -U ADMIN -P admin sdr elist |grep '^OCC'
OCC Active       | 08h | ok  | 210.0 | Device Enabled

Please suggest me if I can submit a patch with the changes mentioned above to the source file op-test-framework/common/OpTestIPMI.py which has the code mentioned in the function ipl_wait_for_working_state().

Or else should we enable the commented line
    #    cmd = 'sdr elist |grep \'Host Status\''
instead of the existing line
        cmd = 'sdr elist |grep \'OCC Active\''

Either way, we need to submit a patch and fix the same in-order to make the Build Verification Tests work on Firestone Hardware.

geissonator, 2015-11-05:
Hey Pavaman, thanks for the detailed info.  I've never been real happy with this portion of the code (we should have a better way to detect a successful boot) but I think what you propose is good for now in that it at least gets the tests working on both systems.  Please fork and submit pull request with this change and I'll merge it in.

PavamanSubramaniyam, 2015-11-18:
I have forked and submitted the pull request with the changes as suggested above.

PavamanSubramaniyam, 2015-11-23:
This issue can be closed since we have merged the pull request with the uncommenting of the original source code lines in the common/OpTestIPMI.py for Host Status sensor checking.

closed 2015-11-23 by PavamanSubramaniyam