open-power/op-test #24: Expand the documentation to talk about coverage

ruscur, 2016-02-03:
Testing an OpenPOWER system can involve a myriad of things, and I don't think it's particularly clear what this repository presently does, or intends to do.  It would be nice to have more documentation, even if just in the README, to cover things like:

- what's being tested?  The AMI BMC?  OpenBMC?  Hostboot, skiboot, petitboot, the kernel?  From what perspective are the tests being ran?  If they're being ran in userspace, what parts of the stack are they touching?

- where are tests being executed?  If there are CI components, is there any publicly visible place they are being run?  https://openpower.xyz comes to mind.

- what hardware is tested?  The example in the README shows a palmetto example, are the tests platform generic?  Could some of them be run in emulators like qemu or simulators like mambo?

It's awesome to see tests, I'm sure I'll be contributing at some point, but I think there needs to be more clarity around what's going on here.

geissonator, 2016-02-10:
I agree, this has been somewhat of a hodge-podge so far.  I'll see if I can put something together.

ping678, 2017-01-25:
I see below info in the op-machines.xsd, who can tell me where I can get the device tree from? I want to know the whole platform list it supports. Thanks.

<xs:annotation><xs:documentation>Specifies what platform this machine is. This MUST appear in the device tree as what type of machine it is. For e
xample: tyan,palmetto or ibm,garrison.</xs:documentation></xs:annotation>

stewart-ibm, 2017-01-30:
The platform name is set in the machine XML, which hostboot adds to the device tree, skiboot uses and ends up being exposed to linux in the root node "compatible" property.

There is no central complete list of all possibilities, as the "ibm,garrison" means that garrison is in the IBM domain. Other vendors are free to use whatever they want (as long as it's under "vendor," prefix).

stewart-ibm, 2017-01-30:
http://github.com/open-power/skiboot/tree/master/platforms is where the platform code lives for skiboot, which acts on specific (and generic) platforms - which is probably the most complete list there is.

stewart-ibm, 2017-03-17:
@ruscur A lot has changed since this issue was originally filed. Does current README help clarify?

ruscur, 2017-03-17:
Yeah it's much better now.  Documentation can always be improved, but I'm happy enough to close this.

closed 2017-03-17 by ruscur