open-power/op-test #34: run-op-bvt command line execution failed, requires inputs to be filled in op-ci-setup.xml file

pridhiviraj, 2016-03-24:
There are couple of issues in executing the framework. As of now command line execution failed, when we gave inputs through command line, It is working only when we fill the inpust in op-ci-setup.xml.

geissonator, 2016-03-24:
Could you please provide more details?  OS, perl, and python versions along with a copy/paste of the failure.

pridhiviraj, 2016-03-24:
Some how in my earlier runs it is failed, then i came to know that we need to include the op-ci-setup.xml file in op-opal-fvt.xml by like below.
<xi:include href="op-ci-setup.xml" parse="xml" xmlns:xi="http://www.w3.org/2001/XInclude"/>
After adding this it is working fine.
this line is not there in op-opal-fvt.xml file. 
So i already send this line in the pull request, #28 

so next time it will work, once it is merged with that change.

Thanks

pridhiviraj, 2016-05-10:
Closing this issue, as @stewart-ibm  fixed the segfault issue and all other issues.

closed 2016-05-10 by pridhiviraj