open-power/op-test #41: Host doesn't have selenium installed error is got

PavamanSubramaniyam, 2016-04-20:
Hi Saqib,

I am trying to execute the tests related to PNOR and BMC Firmware updation using the Web GUI of the BMC with the functionality coded in the OpTestWeb.py which in turn uses the functions in the source files in the directory op-test-framework/common/util/web/.

So, I have developed wrapper scripts in this framework to invoke the functions in OpTestWeb.py
I am getting the below error when I try to execute the tests.

[root@lop826 bvt]# ./run-op-bvt --bmcip 9.40.193.26 --bmcuser sysadmin --bmcpwd superuser --usernameipmi ADMIN --passwordipmi admin --lparip 9.40.193.27 --lparuser root --lparPasswd passw0rd --hpmimage /root/garrison_820.1618.20160414o.hpm --cfgfiledir ../ci/source/ --imagedir /tmp/ --imagename garrison.pnor --fverbose /tmp/new_garrison ./op-bmc-web-update-bvt.xml
01:47:25 :: Command line: ./run-op-bvt --bmcip 9.40.193.26 --bmcuser sysadmin --bmcpwd superuser --usernameipmi ADMIN --passwordipmi admin --lparip 9.40.193.27 --lparuser root --lparPasswd passw0rd --hpmimage /root/garrison_820.1618.20160414o.hpm --cfgfiledir ../ci/source/ --imagedir /tmp/ --imagename garrison.pnor --fverbose /tmp/new_garrison ./op-bmc-web-update-bvt.xml
 26657| 26740|failsumm_log set to: /tmp/root/bvt/2016-04-20-26740/failsumm.log
01:47:25 :: Starting BVT (./op-bmc-web-update-bvt.xml) on 2016-04-20...
01:47:25 :: Log file: /root/run-op-bvt/logs/run-op-bvt.log
 26657| 26740|>getXMLFilePath(./op-bmc-web-update-bvt.xml)
 26657| 26740|   checking for /root/op-test-framework/bvt/./op-bmc-web-update-bvt.xml
 26657| 26740|<getXMLFilePath(./op-bmc-web-update-bvt.xml) returning /root/op-test-framework/bvt/./op-bmc-web-update-bvt.xml
 26657| 26740|fullbvtxmlfile: /root/op-test-framework/bvt/./op-bmc-web-update-bvt.xml
 26657| 26740|run-op-bvt
 26657| 26740|xmlvars{machine}:
 26657| 26740|bvtxmlfile: ./op-bmc-web-update-bvt.xml
 26657| 26740|savedir: /tmp/root/bvt/2016-04-20-26740
 26657| 26740|test_id: op-bmc-web-update   test_state:
 26657| 26740|>runTest(op-bmc-web-update, "OP System BMC and PNOR Firmware update BVT", , op-bmc-web-update.xml)
01:47:25 :: Starting "OP System BMC and PNOR Firmware update BVT" (op-bmc-web-update.xml)
 26657| 26740|   cmd: run-op-it --fverbose  --logpfx op-bmc-web-update --testenv hw   --bmcip 9.40.193.26 --bmcuser sysadmin --bmcpwd superuser --usernameipmi ADMIN --passwordipmi admin --lparip 9.40.193.27 --lparuser root --lparPasswd passw0rd --hpmimage /root/garrison_820.1618.20160414o.hpm --cfgfiledir ../ci/source/ --imagedir /tmp/ --imagename garrison.pnor op-bmc-web-update.xml
01:47:25 :: Starting op-bmc-web-update.xml on 2016-04-20...
01:47:25 :: Starting Test: 1
01:47:25 :: Running command: run-bvt-setup  --bmcip 9.40.193.26 --bmcuser sysadmin --bmcpwd superuser --usernameipmi ADMIN --passwordipmi admin --cfgfiledir ../ci/source/ --ffdcdir %%ffdcdir%% --imagedir /tmp/ --imagename garrison.pnor --lparip 9.40.193.27 --lparuser root --lparPasswd passw0rd --hpmimage /root/garrison_820.1618.20160414o.hpm
01:47:25 :: OK: command finished successfully
01:47:25 :: Finished Test: 1
01:47:25 :: Starting Test: 2
01:47:25 :: Running command: op-ci-bmc-run  "op_bmc_web_update.test_init()"
/root/op-test-framework/ci/source/op_ci_tools.cfg
01:47:25 :: OK: command finished successfully
01:47:25 :: Finished Test: 2
01:47:25 :: Starting Test: 3
01:47:25 :: Running command: op-ci-bmc-run  "op_bmc_web_update.bmc_web_pnor_update_hpm()"
/root/op-test-framework/ci/source/op_ci_tools.cfg
ipmitool -H 9.40.193.26 -I lanplus -U ADMIN -P admin chassis power off
Host doesn't have selenium installed
01:48:20 :: ERROR: command failed with exit code 256
01:48:20 :: ERROR: terminating test
 26657| 26740|   cmd returned 256
01:48:20 :: ERROR: "OP System BMC and PNOR Firmware update BVT" (op-bmc-web-update.xml) finished with one or more errors
 26657| 26740|<runTest(op-bmc-web-update, "OP System BMC and PNOR Firmware update BVT", , op-bmc-web-update.xml)
01:48:20 :: ERROR: Finished BVT tests on 2016-04-20 with 1 testcase failures.
All logs and FFDC saved under /tmp/root/bvt/2016-04-20-26740
 26657| 26740|>run-op-bvt cleanUp(1)
All logs and FFDC saved under /tmp/root/bvt/2016-04-20-26740
 26657| 26740|<run-op-bvt cleanUp(1)

It shows, the error "Host doesn't have selenium installed" which is the code from OpTestConstants.py
ERROR_SELENIUM_HEADLESS = "Host doesn't have selenium installed"
These values are in web_update_hpm function in the source file OpTestWeb.py

So, could you please let me know which all Selenium packages are to be installed on the HOST machines from which I trigger these tests and also the packages required for the target HOST machine.

PavamanSubramaniyam, 2016-04-20:
I am trying to execute the tests from a HOST which has Fedora 22 ppc64le installed. 
It has the below python selenium package installed.
[root@lop826 ~]# rpm -qa | grep selenium
python-selenium-2.45.0-1.fc22.noarch

The target BMC and HOST IPs which are mentioned in the command line parameters is installed with Ubuntu 16.04 ppc64le OS on Garrison Hardware.
root@ltc-garri2:~# dpkg --list | grep python-selenium
ii  python-selenium                                   2.48.0+dfsg1-2                  all          Python bindings for Selenium

saqibkh, 2016-04-20:
You need to install the headless version of selenium on your Host machine i.e. from where you are running the tests (not the BMC/partition).

PavamanSubramaniyam, 2016-04-21:
On my Fedora 22 LE LPAR HOST machine from which I am running the tests,
I have installed the headless version of selenium and followed the steps as mentioned below.

[root@lop826 bvt]# rpm -qa | grep headless
java-1.8.0-openjdk-headless-1.8.0.60-14.b27.fc22.ppc64le

[root@lop826 ~]# pip install selenium
Collecting selenium
  Using cached selenium-2.53.2-py2.py3-none-any.whl
Installing collected packages: selenium
Successfully installed selenium-2.53.2

[root@lop826 ~]# ls -l /usr/lib/python2.7/site-packages/selenium/
total 184
drwxr-xr-x.  2 root root    84 Apr 21 00:55 common
-rw-r--r--.  1 root root   843 Apr 21 00:55 __init__.py
-rw-r--r--.  1 root root   217 Apr 21 00:55 __init__.pyc
-rw-r--r--.  1 root root 78136 Apr 21 00:55 selenium.py
-rw-r--r--.  1 root root 92545 Apr 21 00:55 selenium.pyc
drwxr-xr-x. 14 root root  4096 Apr 21 00:55 webdriver
[root@lop826 ~]# ls -l /usr/lib/python2.7/site-packages/selenium-2.53.2.dist-info/
total 48
-rw-r--r--. 1 root root  4131 Apr 21 00:55 DESCRIPTION.rst
-rw-r--r--. 1 root root     4 Apr 21 00:55 INSTALLER
-rw-r--r--. 1 root root  5051 Apr 21 00:55 METADATA
-rw-r--r--. 1 root root   902 Apr 21 00:55 metadata.json
-rw-r--r--. 1 root root    47 Apr 21 00:55 pbr.json
-rw-r--r--. 1 root root 10908 Apr 21 00:55 RECORD
-rw-r--r--. 1 root root     9 Apr 21 00:55 top_level.txt
-rw-r--r--. 1 root root   110 Apr 21 00:55 WHEEL

Then even downloaded the selenium-server standalone jar file and executed the same

[root@lop826 ~]# DISPLAY=:1 java -jar selenium-server-standalone-2.53.0.jar
01:16:58.268 INFO - Launching a standalone Selenium Server
01:16:58.301 INFO - Java: Oracle Corporation 25.60-b23
01:16:58.301 INFO - OS: Linux 4.0.4-301.fc22.ppc64le ppc64le
01:16:58.313 INFO - v2.53.0, with Core v2.53.0. Built from revision 35ae25b
01:16:58.397 INFO - Driver provider org.openqa.selenium.ie.InternetExplorerDriver registration is skipped:
registration capabilities Capabilities [{ensureCleanSession=true, browserName=internet explorer, version=, platform=WINDOWS}] does not match the current platform LINUX
01:16:58.397 INFO - Driver provider org.openqa.selenium.edge.EdgeDriver registration is skipped:
registration capabilities Capabilities [{browserName=MicrosoftEdge, version=, platform=WINDOWS}] does not match the current platform LINUX
01:16:58.398 INFO - Driver class not found: com.opera.core.systems.OperaDriver
01:16:58.398 INFO - Driver provider com.opera.core.systems.OperaDriver is not registered
01:16:58.400 INFO - Driver provider org.openqa.selenium.safari.SafariDriver registration is skipped:
registration capabilities Capabilities [{browserName=safari, version=, platform=MAC}] does not match the current platform LINUX
01:16:58.401 INFO - Driver class not found: org.openqa.selenium.htmlunit.HtmlUnitDriver
01:16:58.402 INFO - Driver provider org.openqa.selenium.htmlunit.HtmlUnitDriver is not registered
01:16:58.480 INFO - RemoteWebDriver instances should connect to: http://127.0.0.1:4444/wd/hub
01:16:58.481 INFO - Selenium Server is up and running

But still, I get the same error on the HOST from where I am running the tests as:
Host doesn't have selenium installed

Is there any java, python and firefox related packages specific to selenium headless which I would require to install on this HOST.

saqibkh, 2016-04-21:
I have installed the following things on Ubuntu to make it work
0) Update firefox
1) python-pip
2) selenium
3) pyvirtualdisplay

Let me know if you still get the same error by installing the above packages.

PavamanSubramaniyam, 2016-04-25:
I have investigated into the source file common/OpTestWeb.py and found the following issues.

def web_update_hpm(self, i_image, i_component=BMC_CONST.UPDATE_BMCANDPNOR):
        try:
            import argparse
            from selenium import webdriver
            from easyprocess import EasyProcess
            from pyvirtualdisplay import Display
            from FWUpdatePage import FWUpdatePage
            print "I am inside web function"
            from LoginPage import LoginPage

I have tried to put a print debug statement after the importing class functions statement and found that the print statement does not execute at all.
But if the above debugging print statement is placed before the "from FWUpdatePage ........." line in the above source code, then the debug statement gets printed.

Even if I try to make the changes for the importing class functions statement as:
from util.web.FWUpdatePage import FWUpdatePage
then also, the print debug statement does not get executed at all.

Then, I copied __init__.py to the op-test-framework/common/util/ directory and then executed again to see if the print debug works fine, but still the issue persists and the class functions are not getting imported.

Then I have tried to add the importing class function lines outside of  def web_update_hpm ( ) function and before the class OpTestWeb(): line in the source common/OpTestWeb.py.

from OpTestConstants import OpTestConstants as BMC_CONST
from OpTestError import OpTestError

from util.web.FWUpdatePage import FWUpdatePage
from util.web.LoginPage import LoginPage
from util.web.MaintenancePage import MaintenancePage
from util.web.Page import Page

But now I get a new error when I try to execute the scripts.

04:30:50 :: Running command: op-ci-bmc-run  "op_bmc_web_update.test_init()"
Traceback (most recent call last):
  File "<string>", line 7, in <module>
  File "/root/op-test-framework/ci/source/op_ci_bmc.py", line 56, in <module>
    from common.OpTestSystem import OpTestSystem
  File "/root/op-test-framework/common/OpTestSystem.py", line 42, in <module>
    from OpTestWeb import OpTestWeb
  File "/root/op-test-framework/common/OpTestWeb.py", line 36, in <module>
    from util.web.FWUpdatePage import FWUpdatePage
  File "/root/op-test-framework/common/util/web/FWUpdatePage.py", line 31, in <module>
    from OpTestConstants import OpTestConstants as BMC_CONST
ImportError: No module named OpTestConstants

These issues are happening on my Fedora 22 and Ubuntu 16.04 HOSTS from where I am trying to run the tests. The python versions available on my both HOSTS are:

[root@lop826 bvt]# rpm -qf /usr/bin/python
python-2.7.9-6.fc22.ppc64le
[root@lop826 bvt]# python -V
Python 2.7.9

root@hablcpart2048:~/op-test-framework/bvt# dpkg -S /usr/bin/python
python-minimal: /usr/bin/python
root@hablcpart2048:~/op-test-framework/bvt# dpkg --list | grep python-minimal
ii  python-minimal                                    2.7.11-1                                 ppc64el      minimal subset of the Python language (default version)
root@hablcpart2048:~/op-test-framework/bvt# python -V
Python 2.7.11+

PavamanSubramaniyam, 2016-04-25:
From my inference, it seems the classes defined in source files under the util/web/ directory namely FWUpdatePage.py, LoginPage.py, MaintenancePage.py and Page.py are not getting imported into common/OpTestWeb.py functions.

saqibkh, 2016-04-25:
try to use the function below to load the dirs into the sys path before executing the tests(make sure this gets executed first):

full_path = os.path.abspath(os.path.dirname(sys.argv[0])).split('common')[0]
sys.path.append(full_path)
# Walk path and append to sys.path
for root, dirs, files in os.walk(full_path):
    for dir in dirs:
        sys.path.append(os.path.join(root, dir))

PavamanSubramaniyam, 2016-04-26:
I have added the following lines into the file ci/source/op_bmc_web_update.py.

# Get path to base directory and append to path to get common modules
full_path = os.path.dirname(os.path.abspath(__file__))
full_path = full_path.split('ci')[0]
sys.path.append(full_path)

for root, dirs, files in os.walk(full_path):
    for dir in dirs:
        print "the value of dir is"
        print dir
        sys.path.append(os.path.join(root, dir))

Then executed my test cases with the print messages displaying all the directory names

[root@lop826 bvt]# ./run-op-bvt --bmcip 9.3.23.199 --bmcuser sysadmin --bmcpwd superuser --usernameipmi ADMIN --passwordipmi admin --lparip 9.3.29.37 --lparuser root --lparPasswd passw0rd --hpmimage /root/8348_810.1603.20160310b.hpm --cfgfiledir ../ci/source/ --imagedir /tmp/ --imagename  habanero.pnor --verbose --fverbose /tmp/new_habanero ./op-bmc-web-update-bvt.xml
01:20:07 :: Command line: ./run-op-bvt --bmcip 9.3.23.199 --bmcuser sysadmin --bmcpwd superuser --usernameipmi ADMIN --passwordipmi admin --lparip 9.3.29.37 --lparuser root --lparPasswd passw0rd --hpmimage /root/8348_810.1603.20160310b.hpm --cfgfiledir ../ci/source/ --imagedir /tmp/ --imagename habanero.pnor --verbose --fverbose /tmp/new_habanero ./op-bmc-web-update-bvt.xml
  9381|  9784|failsumm_log set to: /tmp/root/bvt/2016-04-26-9784/failsumm.log
01:20:07 :: Starting BVT (./op-bmc-web-update-bvt.xml) on 2016-04-26...
01:20:07 :: Log file: /root/run-op-bvt/logs/run-op-bvt.log
  9381|  9784|>getXMLFilePath(./op-bmc-web-update-bvt.xml)
  9381|  9784|   checking for /root/op-test-framework/bvt/./op-bmc-web-update-bvt.xml
  9381|  9784|<getXMLFilePath(./op-bmc-web-update-bvt.xml) returning /root/op-test-framework/bvt/./op-bmc-web-update-bvt.xml
  9381|  9784|fullbvtxmlfile: /root/op-test-framework/bvt/./op-bmc-web-update-bvt.xml
  9381|  9784|run-op-bvt
  9381|  9784|xmlvars{machine}:
  9381|  9784|bvtxmlfile: ./op-bmc-web-update-bvt.xml
  9381|  9784|savedir: /tmp/root/bvt/2016-04-26-9784
  9381|  9784|test_id: op-bmc-web-update   test_state:
  9381|  9784|>runTest(op-bmc-web-update, "OP System BMC and PNOR Firmware update BVT", , op-bmc-web-update.xml)
01:20:07 :: Starting "OP System BMC and PNOR Firmware update BVT" (op-bmc-web-update.xml)
  9381|  9784|   cmd: run-op-it --fverbose  --logpfx op-bmc-web-update --testenv hw   --bmcip 9.3.23.199 --bmcuser sysadmin --bmcpwd superuser --usernameipmi ADMIN --passwordipmi admin --lparip 9.3.29.37 --lparuser root --lparPasswd passw0rd --hpmimage /root/8348_810.1603.20160310b.hpm --cfgfiledir ../ci/source/ --imagedir /tmp/ --imagename habanero.pnor op-bmc-web-update.xml
01:20:07 :: Starting op-bmc-web-update.xml on 2016-04-26...
01:20:07 :: Starting Test: 1
01:20:07 :: Running command: run-bvt-setup  --bmcip 9.3.23.199 --bmcuser sysadmin --bmcpwd superuser --usernameipmi ADMIN --passwordipmi admin --cfgfiledir ../ci/source/ --ffdcdir %%ffdcdir%% --imagedir /tmp/ --imagename habanero.pnor --lparip 9.3.29.37 --lparuser root --lparPasswd passw0rd --hpmimage /root/8348_810.1603.20160310b.hpm
01:20:07 :: OK: command finished successfully
01:20:07 :: Finished Test: 1
01:20:07 :: Starting Test: 2
01:20:07 :: Running command: op-ci-bmc-run  "op_bmc_web_update.test_init()"
/root/op-test-framework/ci/source/op_ci_tools.cfg
the value of the full path is
/root/op-test-framework/
the value of dir is
.git
the value of dir is
bvt
the value of dir is
ci
the value of dir is
common
the value of dir is
testcases
the value of dir is
refs
the value of dir is
branches
the value of dir is
hooks
the value of dir is
info
the value of dir is
objects
the value of dir is
logs
the value of dir is
heads
the value of dir is
tags
the value of dir is
remotes
the value of dir is
origin
the value of dir is
pack
the value of dir is
info
the value of dir is
refs
the value of dir is
remotes
the value of dir is
heads
the value of dir is
origin
the value of dir is
docs
the value of dir is
%%ffdcdir%%
the value of dir is
source
the value of dir is
build-auto
the value of dir is
util
the value of dir is
web
01:20:08 :: OK: command finished successfully
01:20:08 :: Finished Test: 2
01:20:08 :: Starting Test: 3
01:20:08 :: Running command: op-ci-bmc-run  "op_bmc_web_update.bmc_web_pnor_update_hpm()"
/root/op-test-framework/ci/source/op_ci_tools.cfg
the value of the full path is
/root/op-test-framework/
the value of dir is
.git
the value of dir is
bvt
the value of dir is
ci
the value of dir is
common
the value of dir is
testcases
the value of dir is
refs
the value of dir is
branches
the value of dir is
hooks
the value of dir is
info
the value of dir is
objects
the value of dir is
logs
the value of dir is
heads
the value of dir is
tags
the value of dir is
remotes
the value of dir is
origin
the value of dir is
pack
the value of dir is
info
the value of dir is
refs
the value of dir is
remotes
the value of dir is
heads
the value of dir is
origin
the value of dir is
docs
the value of dir is
%%ffdcdir%%
the value of dir is
source
the value of dir is
build-auto
the value of dir is
util
the value of dir is
web
ipmitool -H 9.3.23.199 -I lanplus -U ADMIN -P admin chassis power of

As can be seen the directories are listed and appended to sys.path

PavamanSubramaniyam, 2016-04-26:
I have observed one more thing in the file common/util/web/LoginPage.py:

from engine.FWObject import FWObject
from connection.common.FWConnection import FWConnection

Since the FWObject and FWConnnection classes are not being used any where in the LoginPage Class, I think we need to remove the above 2 lines.

Also, if we don't comment or remove these lines, we still get "Host doesn't have selenium installed" when we try to execute the tests.

saqibkh, 2016-04-26:
Feel free to remove any unnecessary imports that are causing the failures.

Also it is not much clear from your comment above it you were successfully able to execute the selenium tests. Please confirm!
Thanks

PavamanSubramaniyam, 2016-04-29:
The selenium tests are failing with this message:
Failed to get Maintainance Page
hpm update using webgui failed

In the file common/util/web/MaintenancePage.py ==> 
def getMaintenancePage(self):
        try:
            Maintenance = WebDriverWait(self.Page.driver,BMC_CONST.WEB_DRIVER_WAIT).until(
                       EC.presence_of_element_located((By.ID,
                       BmcPageConstants.BMC_LN_MAINTENANCE)))
            Maintenance.click()

The BMC_LN_MAINTENANCE page is not got due to the presence of element is not located.
EC.presence_of_element_located((By.ID,  ==> is not able to resolve "By.ID" field.

So in the common/OpTestWeb.py
            #Get Maintenance Page
            Maintenance = MaintenancePage(BMC)
            Maintenance.getMaintenancePage()

the Maintenance.getMaintenancePage() is failing and throwing the exception:
"Failed to get Maintainance Page"

saqibkh, 2016-05-03:
Seems that you are passed the selenium package installation.

The problem appears to be in the element id on the maintenance page which probably was updated since the test case was written. Thus needs to be modified

stewart-ibm, 2017-03-17:
So, I don't think this applies anymore with the current tree, closing. We can re-open if the BMC Web UI tests come back (I'm not sure they should though... mostly that should be left to BMC vendors to test)

closed 2017-03-17 by stewart-ibm