open-power/op-test #44: Failed to Check sensors command functionality with different options

SrideviRamesh, 2016-04-26:
Hi,
I have executed test_hwmon_driver() from op-test-framework/testcases/OpTestSensors.py on garrison system. Manually I am able to get output for  "sudo yes | sensors-detect" but when i tried  Checking sensors command functionality with different options... i saw this issue
root@g68L:/home/tyan# sudo sensors -f; echo $?
No sensors found!
Make sure you loaded all the kernel drivers you need.
Try sensors-detect to find out which these are.
1

can you please tell me what are the prerequisites to execute above command

System info:
========
bmc: g68.aus.stglabs.ibm.com
lpar: g68L.aus.stglabs.ibm.com

From BMC:
=========
# cat /proc/ractrends/Helper/FwInfo
FW_VERSION=2.13.00022
FW_DATE=Apr 14 2016
FW_BUILDTIME=16:53:41 CDT
FW_DESC=8335-GTB PASS1 89a207b gar.820.160414a
FW_PRODUCTID=1
FW_RELEASEID=RR9
FW_CODEBASEVERSION=2.X
#

-bash-4.1$  ipmitool -H g68 -I lanplus -U ADMIN -P admin fru print 43
 Product Manufacturer  : 2c80
 Product Name          : 0b
 Product Part Number   : 18KSF1G72PDZ-1G6N1
 Product Version       : 314e
 Product Serial        : 0f5f0f9f
-bash-4.1$

PavamanSubramaniyam, 2016-04-27:
Hi Sridevi,

Can you tell us which is the Linux HOST OS you have installed on Garrisson box.
Have you installed Ubuntu 16.04 or RHEL 7.2 or PKVM as the OS on the machine.

SrideviRamesh, 2016-05-01:
Hi,
I am using
tyan@g68L:~$ cat /etc/os-release
NAME="Ubuntu"
VERSION="15.04 (Vivid Vervet)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 15.04"
VERSION_ID="15.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"

tyan@g68L:~$ cat /boot/config-3.19.0-58-generic | grep -i --color=never CONFIG_SENSORS_IBMPOWERNV
CONFIG_SENSORS_IBMPOWERNV=m

===> 'm' in powerNV OS(i.e RHEL7.2 LE and Ubuntu14.04.04)

PavamanSubramaniyam, 2016-05-02:
We have to load the kernel module manually on your Ubuntu 15.04 HOST OS, since CONFIG_SENSORS_IBMPOWERNV=m is mentioned for this kernel.
So, do the following manually for loading the "ibmpowernv" kernel module.
modprobe ibmpowernv
lsmod | grep ibmpowernv

Then execute the sensors commands with options.

If the kernel config showed CONFIG_SENSORS_IBMPOWERNV=y, then the module would have loaded by default.

SrideviRamesh, 2016-05-02:
Hi pavaman,

"ibmpowernv" is running 

===> This the output of the echo YES | sensor-detect command

root@g68L:/etc/init.d# echo YES | sensors-detect
# sensors-detect revision 6209 (2014-01-14 22:51:58 +0100)
# DMI data unavailable, please consider installing dmidecode 2.7
# or later for better results.

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): modprobe: FATAL: Module cpuid not found.
Failed to load module cpuid.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): Sorry, no supported PCI bus adapters found.

Next adapter: cen_80000004_e0p0 (i2c-0)
Do you want to scan it? (YES/no/selectively): Client found at address 0x1a
Probing for `Analog Devices ADM1021'...                     No
Probing for `Analog Devices ADM1021A/ADM1023'...            No
Probing for `Maxim MAX1617'...                              No
Probing for `Maxim MAX1617A'...                             No
Probing for `Maxim MAX1668'...                              No
Probing for `Maxim MAX1805'...                              No
Probing for `Maxim MAX1989'...                              No
Probing for `Maxim MAX6655/MAX6656'...                      No
Probing for `TI THMC10'...                                  No
Probing for `National Semiconductor LM84'...                No
Probing for `Genesys Logic GL523SM'...                      No
Probing for `Onsemi MC1066'...                              No
Probing for `Maxim MAX1618'...                              No
Probing for `Maxim MAX1619'...                              No
Probing for `National Semiconductor LM82/LM83'...           No
Probing for `Maxim MAX6654'...                              No
Probing for `Maxim MAX6690'...                              No
Probing for `Maxim MAX6680/MAX6681'...                      No
Probing for `Maxim MAX6695/MAX6696'...                      No
Probing for `Texas Instruments AMC6821'...                  No
Probing for `ST STTS424'...                                 No
Probing for `ST STTS424E'...                                No
Probing for `ST STTS2002'...                                No
Probing for `ST STTS3000'...                                No
Probing for `NXP SE97/SE97B'...                             Success!
    (confidence 5, driver `jc42')
Probing for `NXP SE98'...                                   No
Probing for `Analog Devices ADT7408'...                     No
[... 482 lines cut ...]

To load everything that is needed, add this to /etc/modules:
#----cut here----
# Chip drivers
jc42
#----cut here----
If you have some drivers built into your kernel, the list above will
contain too many modules. Skip the appropriate ones!

Do you want to add these lines automatically to /etc/modules? (yes/NO)
***************************************************************
Warning: the preferred way to run this script non-interactively
is with option --auto. Other methods are discouraged and may
stop working at some point in the future.
***************************************************************
===> I did sudo apt-get install lm-sensors and lm-sensors seems to be running
root@g68L:/etc/init.d# ps -eaf | grep lm-sensors
root       3525   3501  0 13:10 pts/0    00:00:00 grep --color=auto lm-sensors

===> but when i run the "sensors; echo $?" command it fails
root@g68L:/etc/init.d# sensors; echo $?
No sensors found!
Make sure you loaded all the kernel drivers you need.
Try sensors-detect to find out which these are.
1

SrideviRamesh, 2016-05-02:
my issue is i am unable to check  sensors command functionality with different options
I doubt that lm-sensors are not properly configured
can u plz share the procedure how to configure lm-sensors

PavamanSubramaniyam, 2016-05-04:
Can you check if you can install Ubuntu 16.04 or 14.04.03 release and try to execute the sensors command, since they are LTS releases and also we had executed these tests on them.

Also, you can try on Ubuntu 15.04 by giving the command:
apt-get install lm-sensors  ==> this will install the lm-sensors package

lm-sensors is not a process which you can verify through "ps -eaf | grep lm-sensors" command as you have tried. This is just a package which has the utility or command sensors-detect, sensors etc.

SrideviRamesh, 2016-05-09:
In my previous comment i have mentioned already that i could do 'apt-get install lm-sensors' ....and sensors seems to be not working....this is my issue

PavamanSubramaniyam, 2016-05-09:
You can raise a defect for the sensors command not working for Ubuntu 15.04 OS release which you have installed here. Then the development team can look into the same and comment.

stewart-ibm, 2017-03-17:
There's been a lot of rework happen in op-test-framework since this was opened, which I believe solves this issue by automatically loading the appropriate drivers if needed.

I'm going to open a separate issue to track the issue of how sensor tests could pass even if no (or too few) sensors are detected.

closed 2017-03-17 by stewart-ibm