open-power/op-test #69: No single command to run full test

stewart-ibm, 2016-08-25:
At best, we can get people to run a single command to run a test suite to confirm conformance.

Currently, the default test suite is not nearly complete enough to ensure that.

geissonator, 2016-08-26:
Yeah, agree, we need a op-conformance-bvt.xml (or something like that) which runs all of our existing test suites that make sense for any Open Power system.

Any volunteers to cook up this xml and validate it against our existing OP systems?

pridhiviraj, 2016-08-26:
yes, i can come up with that xml file and one failure i am seeing is After flashing pnor or hpm image we need to clear BSO, then immediatley all our tests can run. Currently in our ci we are executing standalone script to clear BSO and continuing.
We can made two conformance test suite xml's.
1. Flash PNOR + IPL + Run all existing test suites
2. Flash HPM + IPL + Run all existing test suites

stewart-ibm, 2016-08-29:
We also need the --noflash option to work :)

stewart-ibm, 2017-03-17:
So, I think this has all been addressed in the recent rework, so i'm closing this issue.

closed 2017-03-17 by stewart-ibm