open-power/op-test #85: [open-power: habenaro] i2c driver is non functional in OpenPower Hardware

SrideviRamesh, 2016-11-16:
Explanation of the Issue:

Execution of /op-test-framework/testcases/OpTestI2Cdriver.py failed and test case got hung when executing the command 'sudo i2cdetect -y 6; echo $?' in the test flow

BMC Info

# cat /proc/ractrends/Helper/FwInfo
FW_VERSION=2.16.104497
FW_DATE=Oct 27 2016
FW_BUILDTIME=17:27:08 CDT
FW_DESC=8348 P2 SRC RR9 10272016
FW_PRODUCTID=1
FW_RELEASEID=RR9
FW_CODEBASEVERSION=2.X
#

HOST OS info:

root@habcap10p1:/home/tyan# cat /etc/os-release
NAME="Ubuntu"
VERSION="16.10 (Yakkety Yak)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.10"
VERSION_ID="16.10"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="http://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=yakkety
UBUNTU_CODENAME=yakkety
root@habcap10p1:/home/tyan#

Log info:

Attached opal, dmesg, syslogs collected from the HOST OS
[dmesg_logs.txt](https://github.com/open-power/op-test-framework/files/594369/dmesg_logs.txt)
[opalmsg_logs.txt](https://github.com/open-power/op-test-framework/files/594372/opalmsg_logs.txt)
[sys_logs.txt](https://github.com/open-power/op-test-framework/files/594375/sys_logs.txt)

stewart-ibm, 2017-01-09:
Current code mitigates the issue and does all it can to reset the hardware to a good state, so closing.

stewart-ibm, 2017-01-09:
(only affects systems with a TPM)

closed 2017-01-09 by stewart-ibm