open-power/op-test #96: Support OpenBMC based systems

stewart-ibm, 2017-01-09:
We currently assume IPMI, which is not the case (at least currently) for OpenBMC based systems.

stewart-ibm, 2017-03-17:
We have **very** preliminary support as of https://github.com/open-power/op-test-framework/commit/738db7943eb0f090fa3db07c5200df536d0e3a4a but we need more complete support, notably power on/off/set boot device.

philippe56, 2017-03-22:
In the class OpTestOpenBMC(), shoudn't the ssh port number be 22 (rather than 2200) ?

stewart-ibm, 2017-03-23:
@philippe56 OpenBMCs have 2 SSH ports: 22 is for logging into the BMC itself, and 2200 is for getting access to the host console

stewart-ibm, 2017-03-23:
b19d231096ac27fa137eaf07b22c6deb61032e24 should mostly fix this now that OpenBMC has IPMI.

**However** we still likely have bugs, as the IPMI support in OpenBMC certainly does.

closed 2017-03-23 by stewart-ibm