open-power/op-test #121: Verify DTS sensors

stewart-ibm, 2017-03-23:
We should have several DTS sensors present for each CPU core.

We should verify that these sensors exist for each core as part of the sensors test.

pridhiviraj, 2017-03-23:
This will get us verifying the sensors what we expect in system are created or not. But i am not sure how to validate whether the sensor readings are valid i.e not all zeros and should be within the ranges?

stewart-ibm, 2017-03-23:
Paidipeddi Pridhiviraj <notifications@github.com> writes:
> This will get us verifying the sensors what we expect in system are
> created or not. But i am not sure how to validate whether the sensor
> readings are valid i.e not all zeros and should be within the ranges?

For DTS sensors, there should be a range of sensible values we can look
at. We may have to ask OCC team as to what the absolute limits are.

I can send some mail internally to work that out.

-- 
Stewart Smith
OPAL Architect, IBM.