open-power/op-test #124: Test slot location codes

stewart-ibm, 2017-03-31:
Test that PCI slots have somewhat correct looking slot names.

narasimhan-v, 2017-06-28:
@stewart-ibm Can you explain a bit more here, so that we can implement it.

stewart-ibm, 2017-07-10:
So my thought was we could test a few things to see if slot names were remotely correct (or present).

e.g. on Firestone we have slots with te names "Backplane USB", "Backplane SATA", "Backplane BMC". We could test that slots that have these names have the appropriate devices behind them. e.g. if "Backplane SATA" is actually a network card, we know something is *wrong* (for example, somebody is claiming their machine is a firestone when it obviously isn't). The same with slots named "GPU1"

pridhiviraj, 2018-01-22:
@stewart-ibm #191  implemented for this.

stewart-ibm, 2018-02-22:
Thanks! Now we just have to solve the fact that the tests are failing for some systems/slots.

closed 2018-02-22 by stewart-ibm