open-power/op-test #129: Add testcases for reserved-ranges and reserved-names property

pridhiviraj, 2017-04-03:
Validate the properties against what OPAL actually reserved the regions like size and names.

1. Image Sizes Validation: Verify it from OPAL Message Log

OpenPower systems:
Homer Image     4MB
SLW Image       1MB
OCC Common Area 8MB

cat /sys/firmware/opal/msglog | grep -i size
[   45.078419667,7]   HOMER Image at 0x1ffd800000 size 4MB
[   45.078428921,7]   SLW Image at 0x1ffda00000 size 1MB
[   45.078437864,7]   OCC Common Area at 0x1fff800000 size 8MB

FSP systems:
SLW 1MB
OCC 8MB

cat /sys/firmware/opal/msglog | grep -i size
[   42.301531454,7]   SLW Image at 0x2ffe600000 size 1MB
[   42.301544504,7]   OCC Common Area at 0x2000800000 size 1MB

2. 
Resrved names property
/proc/device-tree# cat reserved-names 
ibm,slw-imageibm,slw-imageibm,slw-imageibm,slw-imageibm,hbrt-code-imageibm,hbrt-target-imageibm,hbrt-vpd-imageibm,firmware-stacksibm,firmware-dataibm,firmware-heapibm,firmware-codeibm,firmware-allocs-memory@2800000000ibm,firmware-allocs-memory@2000000000ibm,firmware-allocs-memory@1000000000ibm,firmware-allocs-memory@0

/proc/device-tree# hexdump -C reserved-ranges 
00000000  00 00 00 2f fe 60 00 00  00 00 00 00 00 10 00 00  |.../�`..........|
00000010  00 00 00 2f fe 20 00 00  00 00 00 00 00 10 00 00  |.../� ..........|
00000020  00 00 00 2f fd e0 00 00  00 00 00 00 00 10 00 00  |.../��..........|
00000030  00 00 00 2f fd a0 00 00  00 00 00 00 00 10 00 00  |.../��..........|
00000040  00 00 00 2f fd 51 00 00  00 00 00 00 00 19 00 00  |.../�Q..........|
00000050  00 00 00 2f fd 6a 00 00  00 00 00 00 00 06 00 00  |.../�j..........|
00000060  00 00 00 2f fd 70 00 00  00 00 00 00 00 10 00 00  |.../�p..........|
00000070  00 00 00 00 31 c0 00 00  00 00 00 00 02 3a 00 00  |....1�.......:..|
00000080  00 00 00 00 31 00 00 00  00 00 00 00 00 c0 00 00  |....1........�..|
00000090  00 00 00 00 30 30 00 00  00 00 00 00 00 d0 00 00  |....00.......�..|
000000a0  00 00 00 00 30 00 00 00  00 00 00 00 00 30 00 00  |....0........0..|
000000b0  00 00 00 28 00 00 00 00  00 00 00 00 00 cc 02 00  |...(.........�..|
000000c0  00 00 00 20 00 00 00 00  00 00 00 00 01 d8 02 00  |... .........�..|
000000d0  00 00 00 10 00 00 00 00  00 00 00 00 00 d8 02 00  |.............�..|
000000e0  00 00 00 00 39 c0 00 00  00 00 00 00 00 d8 02 00  |....9�.......�..|
000000f0

Actual Reserved regions:
[   22.169629796,5] Reserved regions:
[   22.169630777,5]   0x002ffe600000..002ffe6fffff : ibm,slw-image
[   22.169634046,5]   0x002ffe200000..002ffe2fffff : ibm,slw-image
[   22.169637262,5]   0x002ffde00000..002ffdefffff : ibm,slw-image
[   22.169640328,5]   0x002ffda00000..002ffdafffff : ibm,slw-image
[   22.169643433,5]   0x002ffd510000..002ffd69ffff : ibm,hbrt-code-image
[   22.169647093,5]   0x002ffd6a0000..002ffd6fffff : ibm,hbrt-target-image
[   22.169650970,5]   0x002ffd700000..002ffd7fffff : ibm,hbrt-vpd-image
[   22.169654569,5]   0x000031c00000..000033f9ffff : ibm,firmware-stacks
[   22.169658241,5]   0x000031000000..000031bfffff : ibm,firmware-data
[   22.169661620,5]   0x000030300000..000030ffffff : ibm,firmware-heap
[   22.169664945,5]   0x000030000000..0000302fffff : ibm,firmware-code
[   22.169668365,5]   0x002800000000..002800cc01ff : ibm,firmware-allocs-memory@2800000000
[   22.169673871,5]   0x002000000000..002001d801ff : ibm,firmware-allocs-memory@2000000000
[   22.169679361,5]   0x001000000000..001000d801ff : ibm,firmware-allocs-memory@1000000000
[   22.169684887,5]   0x000039c00000..00003a9801ff : ibm,firmware-allocs-memory@0

pridhiviraj, 2017-04-03:
@stewart-ibm  https://github.com/svaidy 
You have any inputs here ?

What hostboot provides, and what opal provides and What we finally reserve the regions?

stewart-ibm, 2017-04-04:
Paidipeddi Pridhiviraj <notifications@github.com> writes:
> @stewart-ibm  https://github.com/svaidy 
> You have any inputs here ?
>
> What hostboot provides, and what opal provides and What we finally
> reserve the regions?

I think it's safe for us to just look at the device tree.

It's possible this test could go in FWTS too.

-- 
Stewart Smith
OPAL Architect, IBM.

pridhiviraj, 2017-04-04:
yeah, that is what i am thinking too, Will add it in fwts, Can you have a list of reserved region names which are fixed in size like homer, slw image?. Other regions(heap and data) will be having varying size.

FSP systems:
SLW 1MB
OCC 8MB

OpenPower systems:
Homer Image 4MB
SLW Image 1MB
OCC Common Area 8MB

I have this list, if you know any other regions are having fixed sizes let me know.

pridhiviraj, 2017-06-28:
closing this issue, as we implemented this in fwts. op-test internally will run fwts. 
https://lists.ubuntu.com/archives/fwts-devel/2017-April/009249.html
https://lists.ubuntu.com/archives/fwts-devel/2017-April/009250.html

closed 2017-06-28 by pridhiviraj