open-power/op-test #144: [RFC] Enabling op-test-framework to deploy distros as part of tests.

harihare, 2017-06-28:
@stewart-ibm  
op-test-framework has all the required backend support to communicate with the openpower system, With the help of these libraries we can easily enable deployment of distros (RHEL, Ubuntu, Centos and Hostos). 
This will help both internal and external teams to use the framework for distro deployment. 

Advantages :
1. We will have more users. 
2. New firmware's can also be tested for kexec with multiple distros and vice-versa.
3. All tests teams internally and externally will be benefited.

sathnaga, 2017-06-28:
This would be a good place as we already have the libraries needed to support it and will be useful for
users incase of host deploy.

stewart-ibm, 2017-06-30:
I've gone on a bit of a spree of not having op-test-framework do any alterations to the machine being tested (at least by default), on the theory that: machines are often shared for development so making changes may not be desirable and that for deployment, better tools are available.

An initial approach would be to add the ability into op-test-framework to select which OS to boot as the OS to do tests with. After that, a next step could be to run the tests in a batch of OSs.

sathnaga, 2017-06-30:
@stewart-ibm Idea is not to change/execute the tests inside the OS, but as initial step having the OS installation as a testcase, I understand there are many opensource tools available for OS installation, 
but op-test framework would help us getting all the required libraries for FSP, BMC and OpenBMC systems handy and it is not going to alter any of the current behaviour of the execution or framework modification, it is just going to an additional set of testcases which would eventually enable us having different host os installed. 

Going forward as you suggested, it can be used to run tests inside that installed OS/available OS in the system if required.

stewart-ibm, 2017-08-16:
Sorry I'm late getting back to this.

So, I think if we had a separate test suite (i.e. `--run-suite os-installers`), and an option to op-test to designate a victim disk (e.g. `--host-victim-disk FOO`) then this could work.

I assume we'd need some kind of webserver to host kickstart files and similar? If so, we should probably work out a way to host those files from op-test itself, and that way we could also host the OS installer images too.

stewart-ibm, 2017-12-04:
https://github.com/open-power/op-test-framework/commit/a3537e347777e087ada5b6de40d43597ba340c10 starts to implement this. It will install Ubuntu.

sathnaga, 2018-02-21:
Below commits refactors the `Ubuntu` install and implements install of `RHEL` and `HostOS`
https://github.com/open-power/op-test-framework/commit/403d141391f5934f548bb1d40049cfd37a020527 https://github.com/open-power/op-test-framework/commit/830a43c6fe54b312a6a63e4c4518c2b72c7b7884 https://github.com/open-power/op-test-framework/commit/c2a96d40fa195413c696a1d7eecdced35617020d https://github.com/open-power/op-test-framework/commit/1b581fbbb61883c7df5de11daccfd734fd9ce1b0 https://github.com/open-power/op-test-framework/commit/4edece0d3c0f3ca239c8bc70fdb8e35e8e99ff23

stewart-ibm, 2018-02-22:
Thanks so much for your work on this!

I think the next step might be to make a distro-install part of the standard test suite if `--scratch-disk` is specified, and we do petitboot magic to boot from it, and then run the 'host' test suite in that distro.

I'll close this issue for now and open a new one for that extra feature though.

closed 2018-02-22 by stewart-ibm