open-power/op-test #145: Need advice on having mvcli tests on petitboot

naveedus, 2017-07-10:
Customers are coming to us to create Marvell RAID VD, so we have working test case implemented for Host OS (RHEL & Ubuntu) in avocado-misc framework. https://github.com/avocado-framework-tests/avocado-misc-tests/pull/564

We would like create a same test case to work on petitboot shell.

Let us know your opinion on having this test case on op-test-framework...

Thank you,
Naveed

naveedus, 2017-07-11:
Hi @stewart-ibm 

We request to have your thoughts on this..

With regards,
Naveed

stewart-ibm, 2017-07-31:
IIRC mvcl is a petitboot plugin?

Having a test could be good, as we certainly do want to ensure that existing plugins continue to work.

Of course, we'll need the test guarded by appropriate checks for a machine that has a Marvell card in it.

naveedus, 2017-08-02:
mvcli is not yet a plugin its just a binary that runs on petiboot shell.
There is a microsemi adaptec raid card which has "arcconf" which comes as plugin form factor. 
Should we ask the development to see if mvcli can be built as pb-plugin?

stewart-ibm, 2017-09-18:
Yeah, I don't want us to start testing running random binaries in petitboot. We support the petitboot plugin interface and the kernel ABI there, and that's the best way to test things.

harihare, 2021-02-04:
As per previous comment, Closing the issue. Please feel free to reopen the issue if you feel it is still relevant.

closed 2021-02-04 by harihare