open-power/op-test #151: Rest API: Change dbus objects from /org/openbmc/ to /xyz/openbmc_project/

pridhiviraj, 2017-07-26:
Currently we have all dbus interfaces for REST API using /org/openbmc this will soon depricated. We need to change it to /xyz/openbmc_project/ . and check existing API is working fine or not.

pridhiviraj, 2017-07-27:
@stewart-ibm We almost changed all the API's to use new /xyz/openbmc_project/ interface in #150 .
Only things to left for TODO is BootProgress and boot_flags. Still they have not yet implemented under /xyz/openbmc_project/

stewart-ibm, 2017-09-18:
```op-test-framework]$ ack '/org/openbmc'
common/OpTestOpenBMC.py
364:    []}' https://bmc/org/openbmc/control/chassis0/action/softPowerOff
368:        obj = "/org/openbmc/control/chassis0/action/softPowerOff"
434:    https://bmc/org/openbmc/records/events/action/clear
438:        obj = "/org/openbmc/records/events/action/clear"
449:    https://bmc/org/openbmc/settings/host0/attr/boot_flags
453:        obj = "/org/openbmc/settings/host0/attr/boot_flags"
460:    https://bmc/org/openbmc/settings/host0/attr/boot_flags
464:        obj = "/org/openbmc/settings/host0/attr/boot_flags"
515:    -X GET https://bmc//org/openbmc/sensors/host/BootProgress
519:        obj = "/org/openbmc/sensors/host/BootProgress"
540:        obj = "/org/openbmc/sensors/host/BootProgress"
```
Still a few to go

pridhiviraj, 2018-01-30:
We are done with replacing all.
```
ack '/org/openbmc/'
common/OpTestOpenBMC.py
583:        obj = "/org/openbmc/sensors/host/BootProgress"
604:        obj = "/org/openbmc/sensors/host/BootProgress"
```
These two are expected fallbacks for standby and runtime checks.

pridhiviraj, 2018-01-30:
So we can close this now.

closed 2018-01-30 by pridhiviraj