open-power/op-test #157: Allow test-reports dir to be specified

thejta, 2017-08-29:
Currently all reports go into a test-reports dir local to the framework install.  That may not always be a writeable space by all calling op-test (if installed on a shared LCB, etc..)

It seems to make sense to give the user an option to specify where the output should go.  -o/--output for the cmdline?

Perhaps the output location could also be controllable via an env variable so the user doesn't have to put it on the cmdline each time - something like OP_TEST_OUTPUT.

Proposed condition flow
If nothing is set: keep current behavior
If OP_TEST_OUTPUT is given and no cmdline: use OP_TEST_OUTPUT
If cmdline is set, regardless of OP_TEST_OUTPUT set or not: use cmdline

The idea sound good?  Any tweaks to switch/variable names?  I'll make the changes, just wanted input before doing so.

stewart-ibm, 2017-09-04:
Jason Albert <notifications@github.com> writes:
> Currently all reports go into a test-reports dir local to the framework install.  That may not always be a writeable space by all calling op-test (if installed on a shared LCB, etc..)
>
> It seems to [...]

All sounds good to me! At some point we'll probably want to generate
better reports and the like (especially as we move into using this for
OpenPOWER Ready and a future compliance suite)

> Proposed condition flow
> If nothing is set: keep current behavior
> If OP_TEST_OUTPUT is given and no cmdline: use OP_TEST_OUTPUT
> If cmdline is set, regardless of OP_TEST_OUTPUT set or not: use [...]

I can't think of anything better than --output, so go with it :)

-- 
Stewart Smith
OPAL Architect, IBM.

thejta, 2017-09-05:
I put in all the changes for this issue in #161, let me know if anything needs tweaking.  I'm not thrilled with the way I print the output location (/dir/*suffix*), so open to ideas there.  I also printed the output path at both the start and end to accommodate making it easier to find logs during longer running tests.

Interesting you mentioned logs.. I created a class in my stuff to run commands and log all output using standard logger.  To do that via xmlrunner, looks like I have to put it to the screen - which isn't user friendly.  If useful to the larger community, happen to contribute it to mainline.

stewart-ibm, 2017-09-18:
Fixed in https://github.com/open-power/op-test-framework/commit/839f0539d0c57aaa1b5436e7a0de1fc3075540dc - closing issue

closed 2017-09-18 by stewart-ibm