open-power/op-test #162: ssl failure in OpTestASM on Ubuntu 14.04

thejta, 2017-09-05:
I get the following error trying to do even a simple -h on 14.04:
`
albertj@opt-ub1404:~/phqf-install$ ./op-test -h
Traceback (most recent call last):
  File "./op-test", line 40, in <module>
    import OpTestConfiguration
  File "/home/albertj/phqf-install/OpTestConfiguration.py", line 8, in <module>
    from common.OpTestFSP import OpTestFSP
  File "/home/albertj/phqf-install/common/OpTestFSP.py", line 40, in <module>
    from OpTestASM import OpTestASM
  File "/home/albertj/phqf-install/common/OpTestASM.py", line 47, in <module>
    ssl._create_default_https_context = ssl._create_unverified_context
AttributeError: 'module' object has no attribute '_create_unverified_context'
`

According to the internet, this wasn't supported until python 2.7.9 and 14.04 is 2.7.6.  https://stackoverflow.com/questions/28228214/ssl-module-object-has-no-attribute-sslcontext

If I comment the line out, everything in my scenarios runs fine - but I don't use the OpTestASM path.  It doesn't seem like there is a easy work around on 2.7.6.  If OpTestASM is not intended to be supported on 14.04, could we put this in a version check to only include on python 2.7.9 or later?

stewart-ibm, 2017-09-18:
What about this:
```
try:
  ssl._create_default_https_context = ssl._create_unverified_context
except AttributeError:
  pass
```

does that silence the error? (or something along those lines)

thejta, 2017-09-18:
I like it, much better than my idea.  And it works!  Tested on ub14.04 and 16.04.  I'll put in the PR.

stewart-ibm, 2017-09-18:
Merged in 189f9336fabd3e71ac72a92e0f5e3d10fc21f3ed, thanks!

closed 2017-09-18 by stewart-ibm