open-power/op-test #192: Detect device tree warnings

ajdlinux, 2018-01-30:
We have a few warnings in the device tree that dtc picks up:

```
Warning (reg_format): "reg" property in /ibm,pcie-slots/root-complex@0,4/switch-up@10b5,8725/down-port@c/builtin@10de,1db1 has invalid length (8 bytes) (#address-cells == 2, #size-cells == 1)
Warning (reg_format): "reg" property in /ibm,pcie-slots/root-complex@0,4/switch-up@10b5,8725/down-port@a/builtin@10de,1db1 has invalid length (8 bytes) (#address-cells == 2, #size-cells == 1)
Warning (reg_format): "reg" property in /ibm,pcie-slots/root-complex@0,4/switch-up@10b5,8725/down-port@b/builtin@10de,1db1 has invalid length (8 bytes) (#address-cells == 2, #size-cells == 1)
```

We could fix all the warnings then add a test for this

pridhiviraj, 2018-01-30:
https://github.com/open-power/op-test-framework/pull/190/commits/310a72783fcba60b74811fe1a37571e5949ef9b2 This test now catches these all DT warnings and errors.

pridhiviraj, 2018-02-02:
@ajdlinux This testcase is merged in https://github.com/open-power/op-test-framework/commit/e25a0632bafb10f6fc7e8b55f274233770d6562c . Can we close this issue now. Thanks.

closed 2018-02-02 by ajdlinux