open-power/op-test #202: Enabling unittest html test report?

sathnaga, 2018-02-21:
It would be helpful if op-test framework can generate html test reports, looks like unittest have capability to generate one.

Sample report: http://tungwaiyip.info/software/sample_test_report.html

sathnaga, 2018-02-21:
Looks like there is a external runner, http://tungwaiyip.info/software/HTMLTestRunner_0_8_2/HTMLTestRunner.py, we might need similar one implemented inside op-test framework.

stewart-ibm, 2018-02-22:
Could we do it like we do the XMLTestRunner today, use it if available but fallback if not?

I also wonder if we can get multiple report formats out...

sathnaga, 2018-02-22:
yes, that will do, we can add those packages to requirements.txt for user aware.
how about having `--html{,xml,...}-report` in command line option through which we can demand framework to generate {html,xml} report?

looks like we have ` html-testRunner` implements, `HtmlTestRunner` as like xmlrunner

[image]

Looks like multiple report format is not possible by unittest framework at the moment(AFAIK), but we can think of an alternative option of converting a test report from one format to another once the test done, by that we can support multiple report formats, not sure how much of it is feasible and a good soultion though...

stewart-ibm, 2018-06-08:
Yeah, perhaps we have to look at upstream python unittest and see if making the changes there is a better idea than us hacking together things.