open-power/op-test #237: console.run_command() fails to execute return value check for certain commands.

sathnaga, 2018-05-11:
Test ran `./op-test -c machine.conf --run testcases.RunHostTest.RunHostTest`

host_cmd_file 
```
cat /proc/cmdline
ls -a
uname -r
```

```
[SOL Session operational.  Use ~? for help]

[console-pexpect]#
[console-pexpect]#^[[H^[[J[console-pexpect]#PS1=[console-pexpect]\#
Shell prompt changed
[console-pexpect]#cat /proc/cmdline
root=UUID=75440ce6-d801-4fc5-b27d-cc3c11dafdc0 ro console=tty0 console=ttyS0,115200 quiet splash crashkernel=2G-4G:320M,4G-32G:512M,32G-64G:1024M,64G-128G:2048M,128G-:4096M@128M
[console-pexpect]#
[console-pexpect]#echo $?
0
# LAST COMMAND EXIT CODE 0 ('0\r\n')
[console-pexpect]#ls -a
.                    .jenkins                               .ssh
..                   .lesshst                               .systemtap
01SV860_151_056.img  linux-diag-ppc64-diag                  .targetcli
.bash_history        .local                                 .vim
.bashrc              .profile                               .viminfo
.cache               root@ltc-test-ci2.aus.stglabs.ibm.com  .wget-hsts
.config              .selected_editor                       .Xauthority
.gnupg               sosreport-guest-20180312015742.tar.xz
[console-pexpect]#
[console-pexpect]#echo $?
0
# LAST COMMAND EXIT CODE 0 ('0\r\n')
[console-pexpect]#uname -r
4.15.0-21-generic
[console-pexpect]#
ERROR (4.939s)

======================================================================
ERROR [4.939s]: runTest (testcases.RunHostTest.RunHostTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins_data/sath/op-test-framework/testcases/RunHostTest.py", line 58, in runTest
    con.run_command(line, timeout=self.host_cmd_timeout)
  File "/home/jenkins_data/sath/op-test-framework/common/OpTestIPMI.py", line 276, in run_command
    exitcode = int(console.before)
ValueError: invalid literal for int() with base 10: ''

----------------------------------------------------------------------
Ran 1 test in 4.939s

FAILED (errors=1)

Generating XML reports...
```

pridhiviraj, 2018-06-05:
This https://github.com/open-power/op-test-framework/pull/241/commits/15e7b12d5e8223b81f1e2cccdb40f1acfc332c7a should fix this issue. @sathnaga Please see if this commit resolved for you.

sathnaga, 2018-06-05:
@pridhiviraj Yep, it fixes the issue, Thanks ....

commit 15e7b12d (2018-06-08): RunHostTest: Fix failure of commands by stripping out the command

Fixes #237

Reported-by: Satheesh Rajendran <sathnaga@linux.vnet.ibm.com>
closed 2018-06-08 by stewart-ibm