open-power/op-test #250: doesn't cope with SSH public key authentication on BMC

ajdlinux, 2018-06-15:
If I install my public key on a BMC, I see errors like the following when using op-test to flash skiboot.

As you can see it uses `-o PubkeyAuthentication=no` when connecting to the box to execute shell commands, but not in the rsync ssh transport, which causes it to get confused when it doesn't see a password prompt.

```
sshpass -p 0penBmc  ssh -p 22 -l root wcapibmc4.aus.stglabs.ibm.com -o PubkeyAuthentication=no -q -o 'UserKnownHostsFile=/dev/null'  -o 'StrictHostKeyChecking=no'
PS1=\[PEXPECT\]#
PS1=\[PEXPECT\]#
root@witherspoon:~# PS1=\[PEXPECT\]#
[PEXPECT]#systemctl restart mboxd.service
[PEXPECT]#echo $?
0
[PEXPECT]#rsync -P -v -e "ssh -k -o StrictHostKeyChecking=no" /home/ajd/code/skiboot/skiboot.lid.xz.stb root@wcapibmc4.aus.stglabs.ibm.com:/tmp
skiboot.lid.xz.stb
        377,240 100%  359.10MB/s    0:00:00 (xfr#1, to-chk=0/1)

sent 2,252 bytes  received 3,269 bytes  1,226.89 bytes/sec
total size is 377,240  speedup is 68.33
ERROR

======================================================================
ERROR: runTest (testcases.OpTestFlash.OpalLidsFLASH)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/ajd/code/op-test-framework/testcases/OpTestFlash.py", line 501, in runTest
    self.cv_BMC.image_transfer(self.skiboot)
  File "/home/ajd/code/op-test-framework/common/OpTestOpenBMC.py", line 1060, in image_transfer
    self.bmc.image_transfer(i_imageName)
  File "/home/ajd/code/op-test-framework/common/OpTestBMC.py", line 139, in image_transfer
    rsync.expect('assword: ')
  File "/usr/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 321, in expect
    timeout, searchwindowsize, async)
  File "/usr/lib/python2.7/dist-packages/pexpect/spawnbase.py", line 345, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python2.7/dist-packages/pexpect/expect.py", line 105, in expect_loop
    return self.eof(e)
  File "/usr/lib/python2.7/dist-packages/pexpect/expect.py", line 50, in eof
    raise EOF(msg)
EOF: End Of File (EOF). Exception style platform.
<pexpect.pty_spawn.spawn object at 0x7f237736b490>
command: /usr/bin/rsync
args: ['/usr/bin/rsync', '-P', '-v', '-e', 'ssh -k -o StrictHostKeyChecking=no', '/home/ajd/code/skiboot/skiboot.lid.xz.stb', 'root@wcapibmc4.aus.stglabs.ibm.com:/tmp']
buffer (last 100 chars): ''
before (last 100 chars): 'ent 2,252 bytes  received 3,269 bytes  1,226.89 bytes/sec\r\ntotal size is 377,240  speedup is 68.33\r\n'
after: <class 'pexpect.exceptions.EOF'>
match: None
match_index: None
exitstatus: None
flag_eof: True
pid: 10625
child_fd: 7
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: <open file '<stdout>', mode 'w' at 0x7f2379d74150>
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile("assword: ")

----------------------------------------------------------------------
Ran 4 tests in 15.997s

FAILED (errors=1, skipped=3)
```

pridhiviraj, 2018-06-18:
@ajdlinux Wondering what is the BMC type you have given in the config. Looks like for an SMC bmc box you have gave OpenBMC as bmc type?

ajdlinux, 2018-06-18:
It's OpenBMC on a Witherspoon.

pridhiviraj, 2018-06-18:
Okay, When BMC is having public key, rsync is copying the file directly w/o asking password.

pridhiviraj, 2018-06-18:
Sent the fix https://github.com/open-power/op-test-framework/pull/254/

closed 2018-06-20 by stewart-ibm