open-power/op-test #267: Make a disk to boot as on option for OS tests

sathnaga, 2018-06-27:
It would be helpful if we can take a optional param as disk and that can be used to boot
and run further tests, will be helpful incase of multiboot system and we can control where we want to run
tests which would help/save time.

there is a method in `common/OpTestInstallUtil.py` - `set_bootable_disk()` but that would require a reboot again, if we can have a handle in petitboot and have a mechanism to boot into specific disk would be more helpful.

Let me know incase if we have that already possible. Thanks.

stewart-ibm, 2018-06-27:
Yeah, I agree that would be useful. It would also be good to be able to pretty easily add kernel parameters as well (e.g. we could then run things with radix_disable and other such options).

@sammj what are your thoughts here?

sammj, 2018-06-28:
Sounds a bit like `--scratch-disk`, except this would be an existing disk rather than installing something fresh right?
I agree also that it would be useful - probably something I can help support via https://github.com/open-power/petitboot/issues/36 too.