open-power/op-test #273: run_DETECT detects the OS state as UNKNOWN.

pridhiviraj, 2018-07-17:
```
[console-expect]#date
Mon Jul 16 05:54:28 EDT 2018
[console-expect]#which whoami && whoami
/usr/bin/whoami
root
[console-expect]#echo $?
0
OpTestSystem now running as root
[console-expect]#
[console-expect]#
[console-expect]#cat /proc/version | cut -d ' ' -f 3 | grep openpower; echo $?
1
[console-expect]#
[console-expect]#
[console-expect]#cat /proc/version | cut -d ' ' -f 3 | grep openpower; echo $?
1
OpTestSystem CURRENT DETECTED STATE: 0
OpTestSystem START STATE: 0 (target 1)
```

debmc, 2018-07-17:
> OpTestSystem CURRENT DETECTED STATE: 0

Did this not work properly ?  If so I can debug.

debmc, 2018-07-17:
> OpTestSystem CURRENT DETECTED STATE: 0

This was as designed, it should have worked properly, but if not let me know.

debmc, 2019-03-29:
@sammj I think this can be closed.

closed 2019-03-31 by sammj