open-power/op-test #275: Simple commands are failing wih exit code -1.

pridhiviraj, 2018-07-17:
```
======================================================================
ERROR [424.678s]: runTest (testcases.OpTestHMIHandling.HMI_TFMR_ERRORS)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pridhiviraj/new_optests/op-test-framework/testcases/OpTestHMIHandling.py", line 483, in runTest
    self._testHMIHandling(BMC_CONST.TFMR_ERRORS)
  File "/home/pridhiviraj/new_optests/op-test-framework/testcases/OpTestHMIHandling.py", line 246, in _testHMIHandling
    self._testTFMR_Errors(BMC_CONST.TFMR_SPURR_PARITY_ERROR)
  File "/home/pridhiviraj/new_optests/op-test-framework/testcases/OpTestHMIHandling.py", line 410, in _testTFMR_Errors
    console.run_command("dmesg -C")
  File "/home/pridhiviraj/new_optests/op-test-framework/common/OpTestIPMI.py", line 312, in run_command
    return self.util.run_command(self, command, timeout, retry)
  File "/home/pridhiviraj/new_optests/op-test-framework/common/OpTestUtil.py", line 541, in run_command
    raise cf
CommandFailed: Command 'dmesg -C' exited with -1.
Output:
['0']

======================================================================
ERROR [11.550s]: runTest (testcases.OpTestHMIHandling.PROC_RECOV_DONE)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pridhiviraj/new_optests/op-test-framework/testcases/OpTestHMIHandling.py", line 496, in runTest
    self._testHMIHandling(BMC_CONST.HMI_PROC_RECV_DONE)
  File "/home/pridhiviraj/new_optests/op-test-framework/testcases/OpTestHMIHandling.py", line 205, in _testHMIHandling
    l_con.run_command("uname -a")
  File "/home/pridhiviraj/new_optests/op-test-framework/common/OpTestIPMI.py", line 312, in run_command
    return self.util.run_command(self, command, timeout, retry)
  File "/home/pridhiviraj/new_optests/op-test-framework/common/OpTestUtil.py", line 541, in run_command
    raise cf
CommandFailed: Command 'uname -a' exited with -1.
Output:
['0']

======================================================================
ERROR [11.143s]: runTest (testcases.OpTestHMIHandling.PROC_RECV_ERROR_MASKED)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pridhiviraj/new_optests/op-test-framework/testcases/OpTestHMIHandling.py", line 500, in runTest
    self._testHMIHandling(BMC_CONST.HMI_PROC_RECV_ERROR_MASKED)
  File "/home/pridhiviraj/new_optests/op-test-framework/testcases/OpTestHMIHandling.py", line 205, in _testHMIHandling
    l_con.run_command("uname -a")
  File "/home/pridhiviraj/new_optests/op-test-framework/common/OpTestIPMI.py", line 312, in run_command
    return self.util.run_command(self, command, timeout, retry)
  File "/home/pridhiviraj/new_optests/op-test-framework/common/OpTestUtil.py", line 541, in run_command
    raise cf
CommandFailed: Command 'uname -a' exited with -1.
Output:
['0']

======================================================================
ERROR [11.147s]: runTest (testcases.OpTestHMIHandling.TOD_ERRORS)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/pridhiviraj/new_optests/op-test-framework/testcases/OpTestHMIHandling.py", line 487, in runTest
    self._testHMIHandling(BMC_CONST.TOD_ERRORS)
  File "/home/pridhiviraj/new_optests/op-test-framework/testcases/OpTestHMIHandling.py", line 205, in _testHMIHandling
    l_con.run_command("uname -a")
  File "/home/pridhiviraj/new_optests/op-test-framework/common/OpTestIPMI.py", line 312, in run_command
    return self.util.run_command(self, command, timeout, retry)
  File "/home/pridhiviraj/new_optests/op-test-framework/common/OpTestUtil.py", line 541, in run_command
    raise cf
CommandFailed: Command 'uname -a' exited with -1.
Output:
['0']

```

stewart-ibm, 2018-07-17:
Possibly related to @debmc recent changes? I haven't seen this myself, so there's something fun going on.

pridhiviraj, 2018-07-17:
@stewart-ibm Yes, it is with latest changes from @debmc .

debmc, 2018-07-17:
> it occurs only in some conditions

Can you provide machine so I can debug ?

pridhiviraj, 2018-07-17:
I think i have fix for it https://github.com/open-power/op-test-framework/pull/276 .
Thanks.

closed 2018-07-18 by stewart-ibm