open-power/op-test #286: op-test-framework warning during flash/boot

hegdevasant, 2018-08-06:
With upstream op-test , I started to see below message even though its making progress.

 *** WaitForIt CURRENT STATE "02" TARGET STATE "06"
 *** WaitForIt working on transition
 *** Current loop iteration "16"             - Reconnect attempts "00" - loop_max "100"
 *** WaitForIt timeout interval "04" seconds - Stale buffer check every "12" times
 *** WaitForIt variables "['/ #', 'Petitboot', 'System shutting down with error status', 'login: ', 'mon> ', 'shutdown requested', 'x=exit']"
 *** WaitForIt Refresh="0" Buffer Kicker="0" - Kill Cord="150"

 *** WaitForIt CURRENT STATE "02" TARGET STATE "06"
 *** WaitForIt working on transition
 *** Current loop iteration "17"             - Reconnect attempts "00" - loop_max "100"
 *** WaitForIt timeout interval "04" seconds - Stale buffer check every "12" times
 *** WaitForIt variables "['/ #', 'Petitboot', 'System shutting down with error status', 'login: ', 'mon> ', 'shutdown requested', 'x=exit']"
 *** WaitForIt Refresh="0" Buffer Kicker="0" - Kill Cord="150"

-Vasant

pridhiviraj, 2018-08-06:
@hegdevasant Yes, these are just informational/debug messages about state transition parameters,. These messages will go to debug log file, when the PR #271 gets in, which actually has the logging framework does logging with two log files(main and debug log files).

hegdevasant, 2018-08-20:
I don't see these warnings with current upstream code. 

Vasant

closed 2018-08-20 by hegdevasant