open-power/op-test #318: [RFC] Support for PowerVM platform (FSP - PHYP based)

harish-24, 2018-10-03:
Adding support for `PowerVM` platform would be very helpful for `PHYP` based Power systems (logical partitions and managed systems).

Currently planning to start with the following with
1. HMC based console support
i.e `mkvterm`,  `rmvterm`
HMC -> LPAR console

2. Handling logical partition/managed system states
i.e `chssysstate`, `lssyscfg`
HMC -> LPAR state

As some of the existing library/APIs could be extended to handle pHYP based machines, it would be a good addition here. Please let know if it could be a value add to the existing `op-test-framework` setup.

stewart-ibm, 2018-10-17:
Interesting... I can see how this could be useful, although I would worry about a bunch of PHYP things creeping into the core code.

One strategy could be the introduction of another system state (along with PETITBOOT, PETITBOOT_SHELL, OS), maybe POWERVM and then any tests requiring that (or able to run in that mode) could just goto_state(POWERVM). Thoughts?

sacsant, 2018-10-20:
We want to keep the PHYP specific things to a bare minimum. Only addition that will be useful from Linux tests point of view is console support and reboot/restart with Linux running in PowerVM LPAR. This will allow us to use op-test as common entry point for all Linux tests (irrespective of the environment -- PowerVM or OPAL).

harish-24, 2018-10-24:
I was thinking more towards not disturbing `OpSystemState` as it has been built around detecting OPAL states. 
It would be better to have a separate state maintenance for the logical partitions' (LPARS) states. That would also help with playing around with LPAR (which is the Host here) shutdown, reboot etc. using HMC.
i.e.
```
class OpLparState():
    NOT_ACTIVE = 'Not Activated'
    RUNNING = 'Running'
    SHUTTING = 'Shutting Down'
    OF = 'Open Firmware'
    STARTING = 'Starting'
    NA = 'Not Available'
```
Controlling and maintaining the LPAR states from OpSystemState would create a overhead as I am guessing it can be better maintained from an independent HMC library. So idea is to have a HMC library maintaining the LPAR states and host console which would also help in isolating POWERVM/PHYP things to the minimum in the existing framework. Please let know your comments on this approach.

closed 2019-07-22 by harish-24