open-power/op-test #346: op-test to detect dracut shell inaddition to xmon for failure boot detection

sathnaga, 2018-11-02:
op-test to detect dracut shell inaddition to xmon for failure boot detection

debmc, 2018-11-02:
> detect dracut shell

if you can provide steps on how to make this happen so I can test out the change I can help.

dracut:/#

debmc, 2018-11-02:
You could also prototype the fix and then submit the patch if its something you can test:

op-test-framework/common/OpTestSystem.py, about line 121 add:

'dracut:/ #'                                    : self.dracut_callback,

OpTestSystem.py around line 226 insert function:

    def dracut_callback(self, **kwargs):
        default_vals = {'my_r': None, 'value': None}
        for key in default_vals:
          if key not in kwargs.keys():
            kwargs[key] = default_vals[key]
        self.state = OpSystemState.UNKNOWN_BAD
        self.stop = 1
        dracut_exception = UnexpectedCase(state=self.state, message="We hit the dracut_callback value={}, manually restart the system".format(kwargs['value']))
        raise guard_exception

sathnaga, 2018-11-19:
Fixed by, https://github.com/open-power/op-test-framework/pull/351

closed 2018-11-23 by stewart-ibm