open-power/op-test #358: How to find out which tests failed?

kyle-ibm, 2018-11-12:
so i was trying out the different test suites and seemed to have gotten a number of tests to run, but i can't tell which tests passed and which tests failed.

in the main.log file it just says there are certain number of failures. 

    ./op-test--config-file zaius1test --run-suite host
and checked results log

    ~/op-test-framework/test-reports/test-run-20181109164756$ cat 20181109084756330511.main.log
    2018-11-09 16:47:56,330:op-test:INFO:Log file: /home/kloh/op-test-framework/test-reports/test-    run-20181109164756/20181109084756330511.main.log
    2018-11-09 16:47:56,330:op-test:INFO:Preparing to set location of Debug Log File to /home/kloh/op-test-framework/test-reports/test-run-20181109164756/20181109084756330945.debug.log
    2018-11-09 16:47:56,331:op-test:INFO:Debug Log file: /home/kloh/op-test-framework/test-reports/test-run-20181109164756/20181109084756330945.debug.log
    2018-11-09 16:47:56,331:op-test:INFO:TestCase Log files: /home/kloh/op-test-framework/test-reports/test-run-20181109164756/*20181109164756*
    2018-11-09 16:47:56,331:op-test:INFO:StreamHandler setup normal
    2018-11-09 16:51:36,186:op-test.common.OpTestUtil:WARNING:OpTestSystem detected something, working on recovery
    2018-11-09 16:51:37,189:op-test.common.OpTestUtil:WARNING:OpTestSystem recovered from a temporary issue, continuing
    2018-11-09 17:01:04,531:op-test:INFO:Exit with Result errors="1" and failures="6"    
    2018-11-09 17:01:04,582:op-test.common.OpTestUtil:INFO  
    ----------------------------------------------------------
    OpTestSystem Firmware Versions Tested
    (if flashed things like skiboot.lid, may not be accurate)
    ----------------------------------------------------------
    Firmware Versions Unavailable
    ----------------------------------------------------------
    -------------------------------------------------------- 

so it say 6 failures but how do i tell what failed?

debmc, 2018-11-12:
Check the following are installed:
pip install unittest-xml-reporting unittest2 xmlrunner

You should see on stdout something like this:
======================================================================
ERROR [0.218s]: runTest (testcases.OpalMsglog.Host)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/debmc/test/testcases/OpalMsglog.py", line 103, in runTest
    "Warnings/Errors in OPAL log:\n%s" % msg)
AssertionError: Warnings/Errors in OPAL log:
[   77.167700152,3] STB: container NOT VERIFIED, resource_id=4 secureboot not yet initialized
[ 1099.192370541,3] MBOX-FLASH: Bad response code from BMC 7
[ 1099.294276623,3] MBOX-FLASH: Error waiting for BMC
[ 1101.117427331,3] MBOX-FLASH: Bad response code from BMC 7
[ 1101.212188361,3] MBOX-FLASH: Error waiting for BMC

======================================================================
ERROR [0.219s]: runTest (testcases.KernelLog.Host)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/debmc/test/testcases/KernelLog.py", line 150, in runTest
    "Warnings/Errors in Kernel log:\n%s" % msg)
AssertionError: Warnings/Errors in Kernel log:
[Mon Nov 12 10:48:49 2018] Crash kernel location must be 0x2000000
[Mon Nov 12 10:48:50 2018] rpadlpar_io_init: partition not DLPAR capable
[Mon Nov 12 10:48:54 2018] pstore: no decompression buffer allocated

stewart-ibm, 2018-11-12:
Yeah, unless you have xmlrunner python module installed you're not going to get anything in the logs about exactly what tests passed/failed.

If you check out the stdout/stderr from the run though, you'll get a text output as to what failed.

kyle-ibm, 2018-11-15:
thanks! getting the xml reports seems to be sufficient for now. closing issue

closed 2018-11-15 by kyle-ibm