open-power/op-test #390: sel/esel dump to separate file

debmc, 2018-12-14:
Currently op-test outputs the sel/esel info to the debug log, it was suggested that having this information in a separate file would be helpful for automation/analysis.

test-reports/elog.txt

hegdevasant, 2019-02-06:
It will be nice if we have test case to download eSEL logs from BMC/FSP and then use tools like opal-elog-parse to get PEL format output.

-Vasant

debmc, 2019-02-06:
> from BMC/FSP

op-test-framework PR #383  https://github.com/open-power/op-test-framework/pull/383

If OpenBMC includes in their capture of info then we can.

nasastry, 2019-04-10:
@debmc PR https://github.com/open-power/op-test-framework/pull/383 got merged. I guess this issue can be closed.

debmc, 2019-04-10:
I think there are different levels of discussion going on here:

This issue deals with just separating out to a new file the already existing sel/esel info op-test-framework already queries from OpenBMC/BMC with existing logic (currently this information is in the op-test-framework debug.log(s).

PR #383 deals with how to use op-test-framework testcase on how to download the OpenBMC dumps to a box (so whatever is in the OpenBMC dump file gets downloaded).  If the OpenBMC dump contains what you want, great.

There is a known issue with OpenBMC NOT including the journal logs:

```
BMC Capture Journal
https://github.com/openbmc/phosphor-debug-collector/issues/6
```

debmc, 2019-04-24:
PR #483 adds this capability.

closed 2019-04-24 by debmc