open-power/op-test #420: New Testcase: Verify Hardware Configuration

dcrowell77, 2019-02-19:
There are a variety of bugs that could result in hardware being hidden from the OS.  This can range from problems in partial good handling of the module vpd, presence detection issues, MRW mistakes, HDAT bugs, etc.  Currently we don't have any tests that ensure we boot with all the parts that the system actually has installed.

The testcase should take in some kind of 'configuration' file that specifies the hardware we expect.  It should include:
- Number of processor chips
- Number of processor cores
- Number of DIMMs and how much memory per DIMM

Theoretically this data could be saved in the AES entry so that it can be fetched dynamically for any system.  Failing that, unique configurations could be maintained for each system in the test pool.

stewart-ibm, 2019-02-20:
The closest thing we currently have is the PCI devices test case, which ensures we get the same `lspci` output as we expect. There also should be one for grabbing it and then, on next IPL, ensure it matches.

At the very least, if we don't have input of what we expect, it should write the data it expects to the test-results/ dir as well as check that on all IPLs we do while testing, we get the same thing we expect.

dcrowell77, 2019-02-20:
Clever idea to just use historical data as the 'right answer'.  That should work as long as the data is manually checked when a new system is added to the pool and it persists forever.