open-power/op-test #428: op-test RHEL install missing REPO and packages

zmoose, 2019-02-25:
I installed RHEL using op-test:
./op-test -c machine.conf --host-scratch-disk /dev/disk/by-id/scsi-35000cca01d0e4d60 --os-repo ftp://partners.redhat.com/1c5d859a/IBM-99752f90e60ef6de524734f8a47c34b1/RHEL-ALT-7.6-GA/Server/ppc64le/os/ --run testcases.InstallRhel.InstallRhel

The installation was successful but there were no REPOs installed and op-test didn't install required packages for our automation. We had requested this be done awhile ago:
https://github.com/open-power/op-test-framework/issues/262

Our DVT automation requires these packages to be installed:
nano 
sysstat
rpm-build
gcc-gfortran
hdparm 
tk
tcsh 
lsof 
python-devel
dtc
libX11-devel
mesa-libGLU-devel
freeglut-devel
ntpdate

sammj, 2019-02-25:
> We had requested this be done awhile ago: #262

Looks like that went in so something must be going wrong in the install. Do you have the test-report output of the install? Is the repo configured in the resulting system?

zmoose, 2019-03-07:
No the test logs are all deleted as we were doing other testing with newer op-test patches.  Currently doing Ubuntu install tests.  Next time we can try RHEL I'll post here with information.

sathnaga, 2019-03-08:
@zmoose 
RHEL install by-default will not setup a repo in the host, i.e something can be add post-process in ks file, but that needs to be more generic as we accept iso aswell for install.

For now the best way can be configured independently using  https://github.com/open-power/op-test-framework/blob/master/testcases/RunHostTest.py, how to use that testcase is documented here, https://sathnaga86.com/2018/11/11/run-host-tests-using-op-test-framework.html

Hope this helps...

harihare, 2021-02-04:
The previous comment addressed the issue.  Hence closing the issue. Please feel free to reopen the bug.

closed 2021-02-04 by harihare