open-power/op-test #435: Move to Python3

stewart-ibm, 2019-03-04:
https://www.python.org/dev/peps/pep-0373/

Says that Python 2.7 will see EOL on Jan 1st 2020.

We should be doing Python 3 well before that.

nasastry, 2019-04-10:
@stewart-ibm 
After going through `official porting guide from python2 to python2/3: https://docs.python.org/3/howto/pyporting.html`

I have tried it on `op-test-framework`, it takes around `+830 -226`.
in the `+830 - if imports are excluded then +304`.

Let me know when to send the patch. The main challenge is testing all the changed test scripts.

nasastry, 2019-04-16:
@stewart-ibm Sent an RFC PR
https://github.com/open-power/op-test-framework/pull/476

oohal, 2019-08-06:
Closed since we've done the conversion.

closed 2019-08-06 by oohal