open-power/op-test #451: Failure to obtain root in Host PNOR tests

amboar, 2019-03-19:
There are some hiomap patches coming down the pipeline that implement alternative backends, and I was attempting to use op-test's OpTestPNOR test cases to exercise the new backend implementation. The host test was failing with the following:

```
======================================================================
ERROR: runTest (testcases.OpTestPNOR.Host)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/andrew/src/open-power/op-test-framework/testcases/OpTestPNOR.py", line 197, in runTest
    self.runTestReadEraseWriteNVRAM()
  File "/home/andrew/src/open-power/op-test-framework/testcases/OpTestPNOR.py", line 106, in runTestReadEraseWriteNVRAM
    self.pflashReadPartition("/tmp/nvram", "NVRAM")
  File "/home/andrew/src/open-power/op-test-framework/testcases/OpTestPNOR.py", line 71, in pflashReadPartition
    self.c.run_command("pflash -r %s -P %s" % (filename,partition))
  File "/home/andrew/src/open-power/op-test-framework/common/OpTestSSH.py", line 211, in run_command
    return self.util.run_command(self, command, timeout, retry)
  File "/home/andrew/src/open-power/op-test-framework/common/OpTestUtil.py", line 1355, in run_command
    raise cf
CommandFailed: Command 'pflash -r /tmp/nvram -P NVRAM' exited with '1'.
Output
["Couldn't initialise architecture flash structures"]
```
This message is printed when `pflash` does not have permission to open the PNOR MTD device. 

The box in question had an `ubuntu` user account, and the `root` account was disabled. op-test was told to use the `ubuntu` user account.

Following up on the lack of priviliege, I found this in the debug log:

```
2019-03-15 15:51:24,931:op-test.common.OpTestUtil:check_root:DEBUG:check_root 2 after=[console-expect]#
 45 2019-03-15 15:51:24,932:op-test.common.OpTestUtil:check_root:DEBUG:check_root echo_rc=0
 44 2019-03-15 15:51:24,932:op-test.common.OpTestUtil:check_root:WARNING:OpTestSystem should be running as root, unable to confirm root, whoami=ubuntu
 43 2019-03-15 15:51:24,933:op-test.common.OpTestSSH:connect:DEBUG:CONNECT starts Expect Buffer ID=0x7f6da752ba10
 42 2019-03-15 15:51:25,258:op-test.common.OpTestSSH:write:DEBUG:uname -a           
 41 2019-03-15 15:51:25,259:op-test.common.OpTestSSH:write:DEBUG:Linux w37l 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:32:48 UTC 2019 ppc64le ppc64le ppc64le GNU/Linux
 40 2019-03-15 15:51:25,259:op-test.common.OpTestSSH:write:DEBUG:[console-expect]#  
 39 2019-03-15 15:51:25,573:op-test.common.OpTestSSH:write:DEBUG:echo $?            
 38 2019-03-15 15:51:25,573:op-test.common.OpTestSSH:write:DEBUG:0                  
 37 2019-03-15 15:51:25,573:op-test.common.OpTestSSH:write:DEBUG:[console-expect]#  
 36 2019-03-15 15:51:25,885:op-test.common.OpTestSSH:write:DEBUG:cat /etc/os-release
 35 2019-03-15 15:51:25,885:op-test.common.OpTestSSH:write:DEBUG:NAME="Ubuntu"      
 34 2019-03-15 15:51:25,886:op-test.common.OpTestSSH:write:DEBUG:VERSION="18.04.2 LTS (Bionic Beaver)"
 33 2019-03-15 15:51:25,886:op-test.common.OpTestSSH:write:DEBUG:ID=ubuntu          
 32 2019-03-15 15:51:25,886:op-test.common.OpTestSSH:write:DEBUG:ID_LIKE=debian     
 31 2019-03-15 15:51:25,886:op-test.common.OpTestSSH:write:DEBUG:PRETTY_NAME="Ubuntu 18.04.2 LTS"
 30 2019-03-15 15:51:25,886:op-test.common.OpTestSSH:write:DEBUG:VERSION_ID="18.04" 
 29 2019-03-15 15:51:25,887:op-test.common.OpTestSSH:write:DEBUG:HOME_URL="https://www.ubuntu.com/"
 28 2019-03-15 15:51:25,887:op-test.common.OpTestSSH:write:DEBUG:SUPPORT_URL="https://help.ubuntu.com/"
 27 2019-03-15 15:51:25,887:op-test.common.OpTestSSH:write:DEBUG:BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
 26 2019-03-15 15:51:25,887:op-test.common.OpTestSSH:write:DEBUG:PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
 25 2019-03-15 15:51:25,887:op-test.common.OpTestSSH:write:DEBUG:VERSION_CODENAME=bionic
 24 2019-03-15 15:51:25,888:op-test.common.OpTestSSH:write:DEBUG:UBUNTU_CODENAME=bionic
 23 2019-03-15 15:51:25,888:op-test.common.OpTestSSH:write:DEBUG:[console-expect]#  
 22 2019-03-15 15:51:26,198:op-test.common.OpTestSSH:write:DEBUG:echo $?            
 21 2019-03-15 15:51:26,199:op-test.common.OpTestSSH:write:DEBUG:0                  
 20 2019-03-15 15:51:26,199:op-test.common.OpTestSSH:write:DEBUG:[console-expect]#  
 19 2019-03-15 15:51:26,512:op-test.common.OpTestSSH:write:DEBUG:pflash -r /tmp/nvram -P NVRAM
 18 2019-03-15 15:51:26,636:op-test.common.OpTestSSH:write:DEBUG:Couldn't initialise architecture flash structures
```
I've started to poke around at the console setup code which should be elevating the session to root using `sudo`, but came away without any real insight beyond there seems to be a lack of separation of responsibilities in the console setup code.

We should be able to run OpTestPNOR against machines with non-root login credentials where the provided user can elevate to root using `sudo`.

amboar, 2019-03-19:
For what it's worth, I ended up working around the problem by giving `root` a password, setting `PermitRootLogin yes` in `/etc/ssh/sshd_config` and then invoking op-build to log into the host as `root`.

sammj, 2019-03-20:
Was concerned this might be related to https://github.com/open-power/op-test-framework/commit/33f9a4b8017b2255f4021d72c1fce16b216c72ca but this also fails with that reverted; must be a rarely exercised path, I'll dig into it

sammj, 2019-03-20:
Actually it was my fault in the end! Should be fixed by https://github.com/open-power/op-test-framework/pull/452

harihare, 2021-02-04:
as per previous comment, closing this issue. Please reopen the issue if you feel still the issue exist.

closed 2021-02-04 by harihare