open-power/op-test #458: Add a test for rebooting the host during a BMC reboot

amboar, 2019-03-26:
The host should be resilient to the BMC being rebooted underneath it even if the host has a dependency on the BMC to access the PNOR.

amboar, 2019-03-26:
I you could assign me @sammj ...

debmc, 2019-03-29:
@amboar There are some tests, not sure exactly what flavor you're looking at, but it may make sense to extend these if applicable.

```
testcases/BasicIPL.py
testcases/BMCResetTorture.py
```

amboar, 2019-03-31:
@debmc thanks, I'll check them out. We have a lingering PNOR/HIOMAP race condition on BMC reboot, so I was thinking of integrating something into OpTestPNOR.

hegdevasant, 2019-04-01:
I have some tests around BMC reboot vs Host/BMC operations [1]. I will respin another version later this week.

[1] https://github.com/open-power/op-test-framework/pull/415

hegdevasant, 2019-08-27:
commit 75d046492e2ff94aafaed425cc7e918af0b613e2
Author: Vasant Hegde <hegdevasant@linux.vnet.ibm.com>
Date:   Mon Feb 11 20:11:16 2019 +0530

    OpenBMC: Add OpenBMC reboot vs Host tests

I think we can close this one.

-Vasant

oohal, 2019-08-27:
ok

closed 2019-08-27 by oohal