open-power/op-test #469: IplParams: Need to update with DD2.3 Risk Levels

debmc, 2019-04-06:
Need to update the testcases/IplParams.py risk level tables with DD2.3 data.

mpe, 2019-04-11:
I believe this is correct for risk level 0:

```
p9_dd2_3_rl0_params = {}
p9_dd2_3_rl0_params["enable"] = ["fw-l1d-thread-split", "inst-l1d-flush-trig2", "inst-spec-barrier-ori31,31,0",
                                 "needs-l1d-flush-msr-hv-1-to-0", "needs-l1d-flush-msr-pr-0-to-1",
                                 "needs-spec-barrier-for-bound-checks", "speculation-policy-favor-security",
                                 "tm-suspend-mode", "fw-branch-hints-honored", "needs-count-cache-flush-on-context-switch",
                                 "fw-count-cache-flush-bcctr2,0,0"]
p9_dd2_3_rl0_params["disable"] = ["inst-l1d-flush-ori30,30,0", "fw-bcctrl-serialized", "inst-thread-reconfig-control-trig0-1",
                                  "fw-ltptr-serialized", "user-mode-branch-speculation", "fw-count-cache-disabled"]
```

I don't know about risk level 1.

debmc, 2019-04-11:
@stewart-ibm Suggestions on where to get the risk level 1 info ?

stewart-ibm, 2019-04-12:
Deb McLemore <notifications@github.com> writes:
> @stewart-ibm Suggestions on where to get the risk level 1 info ?

It gets stashed somewhere in the device tree...

In skiboot hdat/spira.c:

        sys_attributes = be32_to_cpu(p->sys_attributes);
        if (sys_attributes & SYS_ATTR_RISK_LEVEL)
                dt_add_property(node, "elevated-risk-level", NULL, 0);

so should be under /sys-params i think

-- 
Stewart Smith
OPAL Architect, IBM.

debmc, 2019-04-12:
@stewart-ibm  @hegdevasant thanks for the spreadsheet with risk level 1 info, I'll see if I can extract that to build the risk level 1 tables.

debmc, 2019-04-12:
Just an update, I could not get the spreadsheet to sync up with the DD2.3 risk level 0 posted earlier in this issue to even look at the DD2.3 risk level 1 attributes, so I pinged @mpe to take a look at the spreadsheet and advise (I could not get the spreadsheet to upload here).

debmc, 2019-04-12:
An excerpt from an email:

```
P9 DD2.3 in DD2.3 Mode:

root@zz8c-p1:~# ls -1 /proc/device-tree/ibm,opal/fw-features/*/{enabled,disabled}
/proc/device-tree/ibm,opal/fw-features/fw-bcctrl-serialized/disabled
/proc/device-tree/ibm,opal/fw-features/fw-branch-hints-honored/disabled
/proc/device-tree/ibm,opal/fw-features/fw-count-cache-disabled/disabled
/proc/device-tree/ibm,opal/fw-features/fw-count-cache-flush-bcctr2,0,0/enabled
/proc/device-tree/ibm,opal/fw-features/fw-l1d-thread-split/enabled
/proc/device-tree/ibm,opal/fw-features/fw-ltptr-serialized/disabled
/proc/device-tree/ibm,opal/fw-features/inst-l1d-flush-ori30,30,0/disabled
/proc/device-tree/ibm,opal/fw-features/inst-l1d-flush-trig2/enabled
/proc/device-tree/ibm,opal/fw-features/inst-spec-barrier-ori31,31,0/enabled
/proc/device-tree/ibm,opal/fw-features/inst-thread-reconfig-control-trig0-1/disabled
/proc/device-tree/ibm,opal/fw-features/needs-count-cache-flush-on-context-switch/enabled
/proc/device-tree/ibm,opal/fw-features/needs-l1d-flush-msr-hv-1-to-0/enabled
/proc/device-tree/ibm,opal/fw-features/needs-l1d-flush-msr-pr-0-to-1/enabled
/proc/device-tree/ibm,opal/fw-features/needs-pmu-restricted/disabled
/proc/device-tree/ibm,opal/fw-features/needs-spec-barrier-for-bound-checks/enabled
/proc/device-tree/ibm,opal/fw-features/speculation-policy-favor-security/enabled
/proc/device-tree/ibm,opal/fw-features/user-mode-branch-speculation/disabled

```

debmc, 2019-04-16:
Just an update, I chatted with Dean Sanner Apr 15, 2019.

From Dean, it seems that openpower only uses an on/off flag based on the presence of the elevated-risk-level, elevated-risk-level is set if the risk/init level is non 0, so it is by default set if running in DD2.3 mode.

For DD2.3 there are only two -- 4 (Rugby mitigations on) and 5 (no Rugby mitigations)
so for DD2.3 native mode, no Rugby everything would be disabled except for  tm-suspend-mode

```
risk level 0 --> DD2.2 compat mode Kernel + User
risk level 1 --> DD2.2 compat mode Kernel
risk level 2 --> DD2.2 compat mode no Rugby
risk level 4 --> DD2.3 native mode Rugby
risk level 5 --> DD2.3 native mode no Rugby
```

The outstanding question to Dean is "what are the fw-features to be disabled when DD2.3 native mode with elevated-risk-level present (set ON) ?"

Currently there are 15-17+ fw-feature flags so we are working to find the set of fw-features to be validated for DD2.3 native mode.

debmc, 2019-04-20:
Update, chat with Dean Sanner April 17, 2019:

1- Dean will send XLS with init levels defined.
2 - Dean will determine how op-test will find the init level to check against the init levels defined.

debmc, 2019-04-29:
Update, group chat April 29, 2019:

HDAT: Use correct elevated risk value for DD2.3 risk levels
https://w3.rchland.ibm.com/projects/bestquest/?defect=SW464159

Will use -> https://github.com/open-power/hostboot/blob/master/src/usr/hdat/hdatiplparms.H#L520

Risk Level=4 for OpenPOWER Table 0 (no elevated-risk-level)
Risk Level=5 for OpenPOWER Table 1 (elevated-risk-level)

```    def get_params_table(self):
        # P9 Nimbus DD2.3
        # this is HDAT Risk Level=4, which hostboot translates for OpenPOWER
        # if NOT /proc/device-tree/ipl-params/sys-params/elevated-risk-level, use table 0
        # hostboot src/usr/hdat/hdatiplparms.H
        p9_dd2_3_rl0_params = {}
        p9_dd2_3_rl0_params["enable"] = [
                                         "tm-suspend-mode"
                                         "inst-spec-barrier-ori31,31,0",
                                         "inst-l1d-flush-trig2",
                                         "fw-l1d-thread-split",
                                         "fw-count-cache-flush-bcctr2,0,0",
                                         "speculation-policy-favor-security",
                                         "needs-l1d-flush-msr-hv-1-to-0",
                                         "needs-l1d-flush-msr-pr-0-to-1",
                                         "needs-spec-barrier-for-bound-checks",
                                         "needs-count-cache-flush-on-context-switch",
                                         ]
        p9_dd2_3_rl0_params["disable"] = [
                                          "fw-bcctrl-serialized",
                                          "inst-l1d-flush-ori30,30,0",
                                          "fw-branch-hints-honored",
                                          "inst-thread-reconfig-control-trig0-1",
                                          "fw-count-cache-disabled",
                                          "fw-ltptr-serialized",
                                          "user-mode-branch-speculation",
                                          "needs-pmu-restricted",
                                         ]
        # P9 Nimbus DD2.3
        # this is HDAT Risk Level=5, which hostboot translates for OpenPOWER
        # if /proc/device-tree/ipl-params/sys-params/elevated-risk-level, use table 1
        # hostboot src/usr/hdat/hdatiplparms.H
        p9_dd2_3_rl1_params = {}
        p9_dd2_3_rl1_params["enable"] = [
                                         "tm-suspend-mode",
                                        ]
        p9_dd2_3_rl1_params["disable"] = [
                                          "inst-spec-barrier-ori31,31,0",
                                          "inst-l1d-flush-trig2",
                                          "fw-l1d-thread-split",
                                          "fw-count-cache-flush-bcctr2,0,0",
                                          "speculation-policy-favor-security",
                                          "needs-l1d-flush-msr-hv-1-to-0",
                                          "needs-l1d-flush-msr-pr-0-to-1",
                                          "needs-spec-barrier-for-bound-checks",
                                          "needs-count-cache-flush-on-context-switch",
                                          "fw-bcctrl-serialized",
                                          "inst-l1d-flush-ori30,30,0",
                                          "fw-branch-hints-honored",
                                          "inst-thread-reconfig-control-trig0-1",
                                          "fw-count-cache-disabled",
                                          "fw-ltptr-serialized",
                                          "user-mode-branch-speculation",
                                          "needs-pmu-restricted",
                                         ]

```

debmc, 2019-05-03:
Tested with a private PNOR:

https://github.com/open-power/op-test-framework/pull/490

closed 2019-07-27 by debmc