open-power/op-test #495: Falsely identifying system power state during OOBHostLogin test

kyle-ibm, 2019-05-06:
Romulus system was already in Host OS (RHEL 7.6) login screen.
Started OOBHostLogin test and got messages that state system is in standby.
Then upon reboot, script still doesnt detect login prompt correctly and errors out.
----
    $ ./op-test -c rom3test --run testcases.SystemLogin.OOBHostLogin
    Logs in: /home/kloh/op-test-framework/test-reports/test-run-20190506144402
    2019-05-06 14:44:02,050:op-test:setUpLoggerFile:INFO:Preparing to set location of Log File to /home/kloh/op-test-framework/test-reports/test-run-20190506144402/20190506064402050932.main.log
    2019-05-06 14:44:02,051:op-test:setUpLoggerFile:INFO:Log file: /home/kloh/op-test-framework/test-reports/test-run-20190506144402/20190506064402050932.main.log
    2019-05-06 14:44:02,051:op-test:setUpLoggerDebugFile:INFO:Preparing to set location of Debug Log File to /home/kloh/op-test-framework/test-reports/test-run-20190506144402/20190506064402051344.debug.log
    2019-05-06 14:44:02,051:op-test:setUpLoggerDebugFile:INFO:Debug Log file: /home/kloh/op-test-framework/test-reports/test-run-20190506144402/20190506064402051344.debug.log
    2019-05-06 14:44:02,051:op-test:parse_args:INFO:TestCase Log files: /home/kloh/op-test-framework/test-reports/test-run-20190506144402/*
    2019-05-06 14:44:02,052:op-test:parse_args:INFO:StreamHandler setup normal

    Running tests...
    ----------------------------------------------------------------------
      runTest (testcases.OpTestFlash.FSPFWImageFLASH) ... SKIP (0.000s)
      runTest (testcases.OpTestFlash.BmcImageFlash) ... SKIP (0.000s)
      runTest (testcases.OpTestFlash.PNORFLASH) ... SKIP (0.000s)
      runTest (testcases.OpTestFlash.OpalLidsFLASH) ... SKIP (0.000s)

    ----------------------------------------------------------------------
    Ran 4 tests in 0.000s

    OK (skipped=4)

    Generating XML reports...

    Running tests...
    ----------------------------------------------------------------------
      runTest (testcases.SystemLogin.OOBHostLogin) ... command-line line 0: Unsupported option "afstokenpassing"^M
    2019-05-06 14:44:26,294:op-test.common.OpTestUtil:setup_term:INFO:OpTestSystem detected something, checking if your system is powered off, will retry
    2019-05-06 14:44:32,885:op-test.common.OpTestSystem:run_POWERING_OFF:INFO:System is in standby/Soft-off state
    ^L^L^M~.command-line line 0: Unsupported option "afstokenpassing"^M
    ^M~.command-line line 0: Unsupported option "afstokenpassing"^M

    ^M--== Welcome to SBE - CommitId[0x2e253ceb] ==--
    ^Mistep 3.19
    ^Mistep 3.20
    ^Mistep 3.21
    ..
    ..
    ..
    M
    Red Hat Enterprise Linux Server 7.6 (Maipo)
    Kernel 4.14.0-115.el7a.ppc64le on an ppc64le

    rom3p1 login: ^L^M~.^M
    Red Hat Enterprise Linux Server 7.6 (Maipo)
    Kernel 4.14.0-115.el7a.ppc64le on an ppc64le

    rom3p1 login: ERROR (324.072s)

    ======================================================================
    ERROR [324.072s]: runTest (testcases.SystemLogin.OOBHostLogin)
    ----------------------------------------------------------------------
    Traceback (most recent call last):
      File "/home/kloh/op-test-framework/testcases/SystemLogin.py", line 58, in runTest
        r = l_con.run_command("echo 'Hello World'")
      File "/home/kloh/op-test-framework/common/OpTestSSH.py", line 214, in run_command
        return self.util.run_command(self, command, timeout, retry)
      File "/home/kloh/op-test-framework/common/OpTestUtil.py", line 1377, in run_command
        output = self.try_command(term_obj, command, timeout)
      File "/home/kloh/op-test-framework/common/OpTestUtil.py", line 1395, in try_command
        pty = term_obj.get_console() # if previous caller environment leaves buffer hung can show up here, e.g. PS2 prompt
      File "/home/kloh/op-test-framework/common/OpTestSSH.py", line 209, in get_console
        self.check_set_term()
      File "/home/kloh/op-test-framework/common/OpTestSSH.py", line 187, in check_set_term
        self.util.setup_term(self.system, self.pty, None, setup_term_flag)
      File "/home/kloh/op-test-framework/common/OpTestUtil.py", line 1283, in setup_term
        msg="Getting login and sudo not successful, probably connection issue, retry")
    common.Exceptions.ConsoleSettings: Setting the prompt or logging in for the console was not successful, check credentials and review the following for more details
    Expect Before Buffer="^M
    Red Hat Enterprise Linux Server 7.6 (Maipo)
    Kernel 4.14.0-115.el7a.ppc64le on an ppc64le

    rom3p1 login: "
    Expect After Buffer="<class 'pexpect.exceptions.EOF'>" 
    Message="Getting login and sudo not successful, probably connection issue, retry"

    ----------------------------------------------------------------------
    Ran 1 test in 324.072s

    FAILED (errors=1)

    Generating XML reports...
    2019-05-06 14:49:30,337:op-test:<module>:INFO:Exit with Result errors="1" and failures="0"
    2019-05-06 14:49:30,377:op-test.common.OpTestUtil:dump_versions:INFO:Log Location: /home/kloh/op-test-framework/test-reports/test-run-20190506144402/*debug*
    2019-05-06 14:49:30,378:op-test.common.OpTestUtil:dump_versions:INFO:
    ----------------------------------------------------------
    OpTestSystem Firmware Versions Tested
    (if flashed things like skiboot.lid, may not be accurate)
    ----------------------------------------------------------
    None
    ----------------------------------------------------------
    ----------------------------------------------------------
----

Note: I can get the test to work if I press "enter" during the OS login screen (on the SOL console) to reprint the login prompt again in which the script will then catch the expected "login: " 
----

    Running tests...
    ----------------------------------------------------------------------
      runTest (testcases.SystemLogin.OOBHostLogin) ... command-line line 0: Unsupported option "afstokenpassing"^M
    ^M
    Red Hat Enterprise Linux Server 7.6 (Maipo)
    Kernel 4.14.0-115.el7a.ppc64le on an ppc64le

    rom3p1 login: 
    ^M
    Red Hat Enterprise Linux Server 7.6 (Maipo)
    Kernel 4.14.0-115.el7a.ppc64le on an ppc64le

    rom3p1 login: root
    root^M
    Password: xxxxxxxxx

    Last login: Mon May  6 15:48:54 on hvc0
    [root@rom3p1 ~]# which bash && exec bash --norc --noprofile
    PS1=\[console-expect\]#
    which bash && exec bash --norc --noprofile
    /bin/bash
    bash-4.2# PS1=\[console-expect\]#
----

sammj, 2019-05-06:
This should be fixed by https://github.com/open-power/op-test-framework/pull/488

sammj, 2019-05-07:
https://github.com/open-power/op-test-framework/pull/488 has been merged, please retest with the latest op-test master and see if it behaves as expected. :+1:

kyle-ibm, 2019-05-07:
seems to work for OOBHostLogin test but looks like it takes a couple of retries to catch the login phrase. Is that expected?
----
    runTest (testcases.SystemLogin.OOBHostLogin) ... command-line line 0: Unsupported option "afstokenpassing"^M
    2019-05-07 15:06:08,277:op-test.common.OpTestUtil:setup_term:INFO:OpTestSystem detected something, checking if your system is powered off, will retry
    2019-05-07 15:06:33,311:op-test.common.OpTestUtil:setup_term:INFO:OpTestSystem detected something, checking if your system is powered off, will retry
    ^L^L^L
    ^M
    Red Hat Enterprise Linux Server 7.6 (Maipo)
    Kernel 4.14.0-115.el7a.ppc64le on an ppc64le

    rom3p1 login: 
    ^M
    Red Hat Enterprise Linux Server 7.6 (Maipo)
    Kernel 4.14.0-115.el7a.ppc64le on an ppc64le

    rom3p1 login: 
    ^M
    Red Hat Enterprise Linux Server 7.6 (Maipo)
    Kernel 4.14.0-115.el7a.ppc64le on an ppc64le

    rom3p1 login: root
    root^M
    Password: xxxxxxxx
----

sammj, 2019-05-08:
Yes it will retry after trying to redraw with Ctrl-L.

kyle-ibm, 2019-05-10:
got it. closing.

closed 2019-05-10 by kyle-ibm