open-power/op-test #497: OpTestHMIHandling tests fails with "Unknown or new processor type" unless OS is already logged in with the [console-expect]# environment

kyle-ibm, 2019-05-07:
tried running testcases.OpTestHMIHandling.HMI_TFMR_ERRORS with OS booted and waiting in login screen. Login was successful but test failed with
----
    ERROR [65.610s]: runTest (testcases.OpTestHMIHandling.HMI_TFMR_ERRORS)
    ----------------------------------------------------------------------
    Traceback (most recent call last):
      File "/home/kloh/op-test-framework/testcases/OpTestHMIHandling.py", line 622, in runTest
        self._testHMIHandling(BMC_CONST.TFMR_ERRORS)
      File "/home/kloh/op-test-framework/testcases/OpTestHMIHandling.py", line 327, in _testHMIHandling
        self.init_test()
      File "/home/kloh/op-test-framework/testcases/OpTestHMIHandling.py", line 148, in init_test
        l_cores = self.cv_HOST.host_get_cores(console=1)
      File "/home/kloh/op-test-framework/common/OpTestHost.py", line 689, in host_get_cores
        raise OpTestError("Unknown or new processor type")
    common.OpTestError.OpTestError: Unknown or new processor type
----

Then tried running test again but with OS already in the [console-expect]# state and this time, test passed.

I compared the console outputs and it looks like the failing test run never executed the grep revision command which could explain the "Unknown or new processor type" error.

Passing console output
----
    [console-expect]#PATH=/usr/local/sbin:$PATH opal-gard list all
    PATH=/usr/local/sbin:$PATH opal-gard list all
    No GARD entries to display
    [console-expect]#ppc64_cpu --cores-on=all
    ppc64_cpu --cores-on=all
    [console-expect]#echo $?
    echo $?
    0
    [console-expect]#grep '^cpu' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/[,]* .*//;'
    grep '^cpu' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/[,]* .*//;'
    POWER9
    [console-expect]#echo $?
    echo $?
    0
    [console-expect]#grep '^revision' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/ (.*)//;'
    grep '^revision' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/ (.*)//;'
    2.2
    [console-expect]#echo $?
    echo $?
    0
    [console-expect]#grep '^cpu' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/[,]* .*//;'
    grep '^cpu' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/[,]* .*//;'
    POWER9
    [console-expect]#echo $?
    echo $?
    0
    [console-expect]#PATH=/usr/local/sbin:$PATH getscom -l
    PATH=/usr/local/sbin:$PATH getscom -l
    Chip ID  | Rev   | Chip type
    ---------|-------|--------
    00000008 | DD2.2 | P9 (Nimbus) processor
    00000000 | DD2.2 | P9 (Nimbus) processor
    [console-expect]#echo $?
    echo $?
    0
----

Failing console output
----
    console-expect]#PATH=/usr/local/sbin:$PATH opal-gard list all
    PATH=/usr/local/sbin:$PATH opal-gard list all
    No GARD entries to display
    [console-expect]#ppc64_cpu --cores-on=all
    ppc64_cpu --cores-on=all
    [console-expect]#echo $?
    echo $?
    0
    [console-expect]#grep '^cpu' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/[,]* .*//;'
    grep '^cpu' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/[,]* .*//;' ^M^[[A[console-expect]#grep '^cpu' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/[,]* .*//;^[[K'
    POWER9
    [console-expect]#echo $?
    echo $?
    0
    [console-expect]#PATH=/usr/local/sbin:$PATH getscom -l
    PATH=/usr/local/sbin:$PATH getscom -l
    Chip ID  | Rev   | Chip type
    ---------|-------|--------
    00000008 | DD2.2 | P9 (Nimbus) processor
    00000000 | DD2.2 | P9 (Nimbus) processor
    [console-expect]#echo $?
    echo $?
    0
----

sammj, 2019-05-09:
How often does this fail? I've been running it successfully a number of times, with one failure where op-test appeared to fail to properly recognise the return code of a command - perhaps a more general issue.

kyle-ibm, 2019-05-10:
This is repeatable on our Romulus and Zaius systems and on both RHEL7.6 and Ubuntu 18.04.2 SUT host os. The test always fails if i let the script login by itself.
LCB OS is Ubuntu 16.04.5

I did an experiment where i logged in first but did not initiate the [console-expect]# environment and it failed the same way (not executing the grep for revision and thus ending with the Unknown processor type error)

kyle-ibm, 2019-05-17:
after looking at the errors more carefully, I figured out it was erroring out because the host_get_proc_gen method was returning a bad string

for some reason, if the host os wasn't already in the console-expect state, the line returns would be messed up and the host_get_proc_gen would return a long string like this "[console-expect]#grep '^cpu' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/[,]* .*//;'POWER9 " hence eventually causing the unknown processor error.

I hacked the host_get_proc_gen to force a line return and got things to work (though I'm sure yall will have a more elegant solution since this is more of a work around)

----
    def host_get_proc_gen(self, console=0):
        try:
            if self.proc_gen:
                pass
        except AttributeError:
            #self.proc_gen = ''.join(self.host_run_command("grep '^cpu' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/[,]* .*//;'", console=console))
            self.proc_gen = ''.join(self.host_run_command("grep '^cpu' /proc/cpuinfo |uniq|sed -e 's/^.*: //;s/[,]* .*//;';", console=console)).strip(";")
        return self.proc_gen
----

kyle-ibm, 2019-11-21:
haven't seen this issue since cloning a recent op-test version. closing

closed 2019-11-21 by kyle-ibm