open-power/op-test #498: OpTestFastReboot fails to get_petitboot_prompt

debmc, 2019-05-07:
During HB CI, the FastReboot test fails due to verbosity in the fastreboot console output which makes the logic in get_petitboot_prompt not successful.

I've added some more debug logging to HB CI to catch anything helpful, but it seems the logic may need to account for unexpected conditions which may cause the logic to figure out that things may have changed (for instance the petitboot menu data in the buffer was detected but for some reason time passes and we have to be resilient to just detect what we see and then take action based on that).  Anyway, attaching the logs for review and further discussion on how this best can be handled in the future.

Highlevel overview to look for in the logs:

```
2019-05-07 08:47:32,174:op-test.common.OpTestSystem:goto_state:DEBUG:OpTestSystem TRANSITIONED TO: 3
2019-05-07 08:47:32,174:op-test.common.OpTestSystem:petitboot_exit_to_shell:DEBUG:USING PES Expect Buffer ID=0x7fffea329cd0
2019-05-07 08:47:32,225:op-test.common.OpTestSystem:get_petitboot_prompt:DEBUG:USING GPP Expect Buffer ID=0x7fffea329cd0

2019-05-07 08:47:47,608:op-test.common.OpTestSystem:get_petitboot_prompt:DEBUG:USING GPP Expect Buffer ID=0x7fffea329cd0

2019-05-07 08:48:03,688:op-test.common.OpTestSystem:get_petitboot_prompt:DEBUG:USING GPP Expect Buffer ID=0x7fffea329cd0

2019-05-07 08:48:20,206:op-test.common.OpTestSystem:petitboot_exit_to_shell:WARNING:OpTestSystem detected something, tried to recover, but still we have a problem, retry
2019-05-07 08:48:20,211:op-test.common.OpTestSystem:goto_state:DEBUG:OpTestSystem START STATE: 3 (target 4)
2019-05-07 08:48:20,211:op-test.common.OpTestSystem:petitboot_exit_to_shell:DEBUG:USING PES Expect Buffer ID=0x7fffea329cd0
2019-05-07 08:48:20,262:op-test.common.OpTestSystem:get_petitboot_prompt:DEBUG:USING GPP Expect Buffer ID=0x7fffea329cd0

2019-05-07 08:48:40,988:op-test.common.OpTestSystem:get_petitboot_prompt:DEBUG:USING GPP Expect Buffer ID=0x7fffea329cd0

2019-05-07 08:48:48,653:op-test.common.OpTestSSH:write:DEBUG:[  615.459250590,3] I2C: Failed to program the MODE_REG
2019-05-07 08:48:48,653:op-test.common.OpTestSSH:write:DEBUG:
2019-05-07 08:48:48,653:op-test.common.OpTestSSH:write:DEBUG: Running /bin/sh...^[[K^[[H^[[J^[[24;1H^[[?25h^[[?0c
2019-05-07 08:48:48,653:op-test.common.OpTestSSH:write:DEBUG:^[c^[(B^[[m^[[J^[[?25hExiting petitboot. Type 'exit' to return.
2019-05-07 08:48:48,653:op-test.common.OpTestSSH:write:DEBUG:You may run 'pb-sos' to gather diagnostic data
2019-05-07 08:48:48,653:op-test.common.OpTestSSH:write:DEBUG:No password set, running as root. You may set a password in the System Configuration screen.
2019-05-07 08:48:48,654:op-test.common.OpTestSSH:write:DEBUG:#
2019-05-07 08:48:48,654:op-test.common.OpTestSSH:write:DEBUG:# x
2019-05-07 08:48:48,654:op-test.common.OpTestSSH:write:DEBUG:-sh: x: not found
2019-05-07 08:48:48,654:op-test.common.OpTestSSH:write:DEBUG:# x
2019-05-07 08:48:48,654:op-test.common.OpTestSSH:write:DEBUG:-sh: x: not found
2019-05-07 08:48:48,654:op-test.common.OpTestSSH:write:DEBUG:# x
2019-05-07 08:48:48,654:op-test.common.OpTestSSH:write:DEBUG:-sh: x: not found
2019-05-07 08:48:48,654:op-test.common.OpTestSSH:write:DEBUG:# x
2019-05-07 08:48:48,654:op-test.common.OpTestSSH:write:DEBUG:-sh: x: not found
2019-05-07 08:48:48,655:op-test.common.OpTestSSH:write:DEBUG:#

```

debmc, 2019-05-07:
[179-pnor01-skiroot-20190507133607955243.debug.log.5.txt](https://github.com/open-power/op-test-framework/files/3153157/179-pnor01-skiroot-20190507133607955243.debug.log.5.txt)

debmc, 2019-05-07:
[179-pnor01-skiroot-20190507133607955243.debug.log.4.txt](https://github.com/open-power/op-test-framework/files/3153158/179-pnor01-skiroot-20190507133607955243.debug.log.4.txt)

debmc, 2019-05-07:
[179-pnor01-skiroot-20190507133607955243.debug.log.3.txt](https://github.com/open-power/op-test-framework/files/3153163/179-pnor01-skiroot-20190507133607955243.debug.log.3.txt)

debmc, 2019-05-07:
[179-pnor01-skiroot-20190507133607955243.debug.log.2.txt](https://github.com/open-power/op-test-framework/files/3153164/179-pnor01-skiroot-20190507133607955243.debug.log.2.txt)

debmc, 2019-05-07:
[179-pnor01-skiroot-20190507133607955243.debug.log.1.txt](https://github.com/open-power/op-test-framework/files/3153166/179-pnor01-skiroot-20190507133607955243.debug.log.1.txt)

debmc, 2019-05-07:
[179-pnor01-skiroot-20190507133607955243.debug.log.txt](https://github.com/open-power/op-test-framework/files/3153167/179-pnor01-skiroot-20190507133607955243.debug.log.txt)

harihare, 2021-02-04:
There was a recent fix for fastreboot issue https://github.com/open-power/op-test/pull/609
Please retry and reopen the issue if it is still not fixed.

closed 2021-02-04 by harihare