open-power/op-test #543: Re-factoring PowerVM implementation to adopt existing model

harish-24, 2019-08-30:
The op-test framework works with different BMC types, manages various consoles, system control through `OpTestSystem` class which maintains machine [states](https://github.com/open-power/op-test/blob/master/common/OpTestSystem.py#L58) including OS, PETITBOOT, OFF etc.

The current implementation of HMC interface does not leverage the `OpTestSystem` class  for accessing console ( `self.cv_SYSTEM.console` ) or managing system (`self.cv_SYSTEM.goto_state()`) from a testcase perspective. 
Instead HMC uses only a `self.hmc` object of the HMC class to avoid touching the existing system states and management. (RFC: #318)

To comply common usage from tests, HMC has to manage the system scanning through the existing system states which is different for an LPAR ([OpHmcState](https://github.com/open-power/op-test/blob/master/common/OpTestHMC.py#L48) and [OpManagedState](https://github.com/open-power/op-test/blob/master/common/OpTestHMC.py#L61)). 

Possible solution could be
1. To integrate these states as system states in `OpTestSystem` class. And if so, handle all functions to adopt to these states to leverage common usage.

2. Provide a wrapper for the around HMC object from OpTestSystem and override the function similar to existing model and do not touch the existing functions.

It would help us better if you could share your thoughts on this for better implementation and handling which would help everyone eventually.

oohal, 2019-09-02:
> The op-test framework works with different BMC types, manages various consoles, system control through `OpTestSystem` class which maintains machine [...]

Avoiding touching the core stuff might seem like a good idea, but the end result is that it's moved responsibility for handling platform differences into the individual tests. Not really a win IMO since the whole point of op-test was to provide a system independent framework for running tests.

> To comply common usage from tests, HMC has to manage the system scanning through the existing system states which is different for an LPAR [...]

The state of the managed system is an implementation detail on the LPAR backend. You might write one or two tests that require power cycling the managed system, but for the most part a test is only interested in starting or stopping an individual LPAR. With that in mind the actual states we care about boil down to:

```
Off
Starting
In the boot loader
In the OS
Powering off
```
The names are slightly different between powernv and an LPAR, but the broad strokes are the same.

> Possible solution could be
>
>     1. To integrate these states as system states in `OpTestSystem` class. And if so, handle all functions to adopt to these states to leverage common usage.
>
>     2. [...]

Do both IMO. Fundementally op-test has two real jobs:

a) Turn the system under test on and off
b) Get a console.

OpTestSystem should really just be the API used by testcases to access these functions. Other, more platform specific, behaviour should come from the backend. We already sort of do that with the various feature-test functions in the (e.g. has_inband_bootdev()), but we could do a much better job of it.

The other part is to introduce platform-specific sub-states that a test can request. This would be where things like PETITBOOT_SHELL (maybe GRUB_SHELL on LPAR?) would go so that tests can still get to these states if they need them, but when testing generic linux features (e.g. Kdump) the test doesn't need to care.

Hope that clears things up a bit.

harish-24, 2019-09-03:
> 
> Avoiding touching the core stuff might seem like a good idea, but the end result is that it's moved responsibility for handling platform differences into the individual tests. Not really a win IMO since the whole point of op-test was to provide a system independent framework for running tests.
> 
Sure, Agreed. 

> The state of the managed system is an implementation detail on the LPAR backend. You might write one or two tests that require power cycling the managed system, but for the most part a test is only [...]

Just  to make it clear, the existing states are exactly what is being defined from actual HMC interface. You mean we need to define different states to match OpSystemState?

> 
> 
> Do both IMO. Fundementally op-test has two real jobs:
> 
> a) Turn the system under test on and off
> b) Get a console.
> 
> OpTestSystem should really just be the API used by testcases to access these functions. Other, more platform specific, behaviour should come from the backend. We already sort of do that with the various feature-test functions in the (e.g. has_inband_bootdev()), but we could do a much better job of it.
> 
> Hope that clears things up a bit.

Thanks for all the inputs. To be clear is the following good to proceed with?
Provide a back-end support from a `OpTestSystem` to map to LPAR states/functions and make use of same OpTestSystem interface from any generic tests?

oohal, 2019-09-04:
yeah, that sounds about right

PR #555 (2019-11-28): [v2] Refactor PowerVM code aligning to existing framework
This patch refactors existing PowerVM specific code implementation to be aligned with the other BMC types i.e. irrespective of the BMC type.

An LPAR is considered a SUT equivalent to a PowerNV machine.

This patch helps us achieve a common interface of accessing and using SUT (LPAR or PowerNV). Methods which are not applicable for LPARs are overridden wherever possible.

v2 Changelog:
1) Get console connected initially for test to run commands with
2) Detect initial poweroff and recover system for use

Fixes: #543
commit 386436a5 (2019-11-28): [v2] Refactor PowerVM code aligning to existing framework

This patch refactors existing PowerVM specific code implementation
to be aligned with the other BMC types i.e. irrespective of the
BMC type.

An LPAR is considered a SUT equivalent to a PowerNV machine.

This patch helps us achieve a common interface of accessing and
using SUT (LPAR or PowerNV). Methods which are not applicable for
LPARs are overridden wherever possible.

v2 Changelog:
1) Get console connected initially for test to run commands with
2) Detect initial poweroff and recover system for use

Fixes: #543
closed 2019-11-28 by oohal