open-power/op-test #587: InstallUpstreamKernel.py: getting ssh console was unsuccessful some times

abdhaleegit, 2020-05-06:
```
[   32.081422] IPv6: ADDRCONF(NETDEV_CHANGE): vnic1: link becomes ready^M
[   34.755227] IPv6: ADDRCONF(NETDEV_CHANGE): vnic2: link becomes ready^M
^M^M
Red Hat Enterprise Linux 8.1 (Ootpa)^M
Kernel 5.7.0-rc4+ on an ppc64le^M
^M
Activate the web console with: systemctl enable --now cockpit.socket^M
^M
ltc-zz1b-lp4 login: [console-expect]#command-line line 0: Unsupported option "afstokenpassing"^M

ERROR

======================================================================
ERROR: runTest (testcases.InstallUpstreamKernel.InstallUpstreamKernel)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins/jenkins_home/workspace/ioci/testcases/InstallUpstreamKernel.py", line 150, in runTest
    res = con.run_command("uname -r")
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestSSH.py", line 225, in run_command
    return self.util.run_command(self, command, timeout, retry)
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestUtil.py", line 1611, in run_command
    output = self.try_command(term_obj, command, timeout)
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestUtil.py", line 1631, in try_command
    pty = term_obj.get_console()
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestSSH.py", line 205, in get_console
    self.connect(logger)
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestSSH.py", line 185, in connect
    self.check_set_term()
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestSSH.py", line 201, in check_set_term
    self, setup_term_flag)
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestUtil.py", line 1506, in setup_term
    msg="Getting login and sudo not successful, probably connection issue, retry")
common.Exceptions.ConsoleSettings: Setting the prompt or logging in for the console was not successful, check credentials and review the following for more details
Expect Before Buffer="command-line line 0: Unsupported option "afstokenpassing"^M^M
"
Expect After Buffer="<class 'pexpect.exceptions.EOF'>"
Message="Getting login and sudo not successful, probably connection issue, retry"

----------------------------------------------------------------------
Ran 1 test in 1584.104s

FAILED (errors=1)
2020-05-05 18:22:30,565:op-test:<module>:INFO:Exit with Result errors="1" and failures="0"
2020-05-05 18:22:30,566:op-test.common.OpTestUtil:cleanup:INFO:OpTestSystem Starting to Gather ESEL's
2020-05-05 18:22:30,566:op-test.common.OpTestUtil:dump_versions:INFO:Log Location: /home/jenkins/jenkins_home/workspace/ioci/test-reports/test-run-20200505175528/*debug*
2020-05-05 18:22:30,567:op-test.common.OpTestUtil:dump_versions:INFO:

```

harihare, 2020-05-22:
Looking into the issue.

harihare, 2020-05-26:
Tried Installation  on OpenBMC box, it ran successfully. 
./op-test -c ./../config/wspoon2.cnf --run testcases.InstallUpstreamKernel.InstallUpstreamKernel
System was installed with the upstream kernel.  
Discussed with @abdhaleegit he mentioned it happens every 10 tries on PowerVM. Will check on VM too.

harihare, 2020-05-27:
Was able to recreate the issue on PowerVM, looking into the issue.

abdhaleegit, 2020-06-30:
@harihare I am seeing this issue at different stages of the test here is the new trace

```
[console-expect]#echo $?
echo $?
0
[console-expect]#^M~.

Red Hat Enterprise Linux 8.1 (Ootpa)
Kernel 4.18.0-147.el8.ppc64le on an ppc64le

Activate the web console with: systemctl enable --now cockpit.socket

ltc-zz1b-lp7 login: ERROR

======================================================================
ERROR: runTest (testcases.InstallUpstreamKernel.InstallUpstreamKernel)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins/jenkins_home/workspace/ioci@2/testcases/InstallUpstreamKernel.py", line 147, in runTest
    raw_pty = self.cv_SYSTEM.console.get_console()
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestHMC.py", line 487, in get_console
    None, self.system.block_setup_term)
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestUtil.py", line 1506, in setup_term
    msg="Getting login and sudo not successful, probably connection issue, retry")
common.Exceptions.ConsoleSettings: Setting the prompt or logging in for the console was not successful, check credentials and review the following for more details
Expect Before Buffer="Red Hat Enterprise Linux 8.1 (Ootpa)
Kernel 4.18.0-147.el8.ppc64le on an ppc64le

Activate the web console with: systemctl enable --now cockpit.socket

ltc-zz1b-lp7 login: "
Expect After Buffer="<class 'pexpect.exceptions.TIMEOUT'>" 
Message="Getting login and sudo not successful, probably connection issue, retry"

abdhaleegit, 2020-06-30:
in another system i see this

```
console-expect]#echo $?
echo $?
0
[console-expect]#^M~.

Red Hat Enterprise Linux 8.1 (Ootpa)
Kernel 4.18.0-147.el8.ppc64le on an ppc64le

Activate the web console with: systemctl enable --now cockpit.socket

ltc-zz1b-lp2 login: 

Red Hat Enterprise Linux 8.1 (Ootpa)
Kernel 4.18.0-147.el8.ppc64le on an ppc64le

Activate the web console with: systemctl enable --now cockpit.socket

ltc-zz1b-lp2 login: root

root

2020-06-29 18:11:58,707:op-test.common.OpTestUtil:get_login:WARNING:OpTestSystem Problem with the login and/or password prompt, raised Exception ConsoleSettings but continuing
ERROR

======================================================================
ERROR: runTest (testcases.InstallUpstreamKernel.InstallUpstreamKernel)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins/jenkins_home/workspace/ioci@2/testcases/InstallUpstreamKernel.py", line 147, in runTest
    raw_pty = self.cv_SYSTEM.console.get_console()
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestHMC.py", line 487, in get_console
    None, self.system.block_setup_term)
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestUtil.py", line 1490, in setup_term
    system_obj.cv_HOST, term_obj, pty, self.build_prompt(system_obj.prompt))
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestUtil.py", line 1299, in get_login
    msg="Problem with the login and/or password prompt, probably a connection or credential issue, retry")
common.Exceptions.ConsoleSettings: Setting the prompt or logging in for the console was not successful, check credentials and review the following for more details
Expect Before Buffer="root

"
Expect After Buffer="<class 'pexpect.exceptions.TIMEOUT'>" 
Message="Problem with the login and/or password prompt, probably a connection or credential issue, retry"

----------------------------------------------------------------------
Ran 1 test in 1342.156s

FAILED (errors=1)
2020-06-29 18:12:00,712:op-test:<module>:INFO:Exit with Result errors="1" and failures="0"
2020-06-29 18:12:00,712:op-test.common.OpTestUtil:cleanup:INFO:OpTestSystem Starting to Gather ESEL's
2020-06-29 18:12:00,713:op-test.common.OpTestUtil:dump_versions:INFO:Log Location: /home/jenkins/jenkins_home/workspace/ioci@2/test-reports/test-run-20200629174900/*debug*
2020-06-29 18:12:00,713:op-test.common.OpTestUtil:dump_versions:INFO:

abdhaleegit, 2020-07-06:
@harihare We are seeing this more often, can you please provide some comments if any thing from your debugging ?

abdhaleegit, 2020-07-08:
The below patch fixes the issue, will be sending out the same 
```
diff --git a/common/OpTestUtil.py b/common/OpTestUtil.py
index 162b8ce..34e6ce0 100644
--- a/common/OpTestUtil.py
+++ b/common/OpTestUtil.py
@@ -1454,7 +1454,7 @@ class OpTestUtil():
             return
         rc = pty.expect(['login: $', ".*#$", ".*# $", ".*\$", "~>", "~ #",
-                         'Petitboot', pexpect.TIMEOUT, pexpect.EOF], timeout=10)
+                         'Petitboot', pexpect.TIMEOUT, pexpect.EOF], timeout=100)
         if rc == 0:
             track_obj.PS1_set, track_obj.LOGIN_set = self.get_login(
                 system_obj.cv_HOST, term_obj, pty, self.build_prompt(system_obj.prompt))

abdhaleegit, 2020-07-13:
looks like we hit it again with different code path

```
[[0;32m  OK  [0m] Started LSB: Starts and stops the InfiniBand SRP client service.

Red Hat Enterprise Linux 8.1 Beta (Ootpa)
Kernel 4.18.0-144.el8.ppc64le on an ppc64le

Activate the web console with: systemctl enable --now cockpit.socket

ltc-zz3-lp1 login: 5

5
Password: [console-expect]#command-line line 0: Unsupported option "afstokenpassing"^M
login: timed out after 60 seconds

Red Hat Enterprise Linux 8.1 Beta (Ootpa)
Kernel 4.18.0-144.el8.ppc64le on an ppc64le

Activate the web console with: systemctl enable --now cockpit.socket

ltc-zz3-lp1 login: 

^M~.
~.command-line line 0: Unsupported option "afstokenpassing"^M

^M~.
ERROR

======================================================================
ERROR: runTest (testcases.InstallUpstreamKernel.InstallUpstreamKernel)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins/jenkins_home/workspace/ioci@2/testcases/InstallUpstreamKernel.py", line 92, in runTest
    (self.home, self.home))
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestSSH.py", line 225, in run_command
    return self.util.run_command(self, command, timeout, retry)
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestUtil.py", line 1611, in run_command
    output = self.try_command(term_obj, command, timeout)
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestUtil.py", line 1631, in try_command
    pty = term_obj.get_console()
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestSSH.py", line 205, in get_console
    self.connect(logger)
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestSSH.py", line 185, in connect
    self.check_set_term()
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestSSH.py", line 201, in check_set_term
    self, setup_term_flag)
  File "/home/jenkins/jenkins_home/workspace/ioci@2/common/OpTestUtil.py", line 1506, in setup_term
    msg="Getting login and sudo not successful, probably connection issue, retry")
common.Exceptions.ConsoleSettings: Setting the prompt or logging in for the console was not successful, check credentials and review the following for more details
Expect Before Buffer="command-line line 0: Unsupported option "afstokenpassing"

~."
Expect After Buffer="<class 'pexpect.exceptions.TIMEOUT'>" 
Message="Getting login and sudo not successful, probably connection issue, retry"

abdhaleegit, 2020-07-13:
@harihare can you have a look at it ?

klauskiwi, 2020-07-24:
@abdhaleegit so just increasing the timeout solves the issue? I think increasing it 10-fold might be excessive, but in your testing, can you tell how many seconds would be acceptable? Can you send a PR? Thanks

abdhaleegit, 2021-01-18:
@klauskiwi just increasing the timeout did not solve the problem.. looks like there is more to it. I able to see the problem even with increase timeout. @harihare I am seeing consistently on  p10 lpars

```
2021-01-13 07:57:14,270:op-test.common.OpTestHMC:write:DEBUG:Red Hat Enterprise Linux 8.3 (Ootpa)
2021-01-13 07:57:14,270:op-test.common.OpTestHMC:write:DEBUG:Kernel 5.11.0-rc2+ on an ppc64le
2021-01-13 07:57:14,270:op-test.common.OpTestHMC:write:DEBUG:
2021-01-13 07:57:14,271:op-test.common.OpTestHMC:write:DEBUG:Activate the web console with:
2021-01-13 07:57:14,273:op-test.common.OpTestHMC:write:DEBUG:systemctl enable --now cockpit.socket
2021-01-13 07:57:14,273:op-test.common.OpTestHMC:write:DEBUG:
2021-01-13 07:57:14,273:op-test.common.OpTestHMC:write:DEBUG:ltcden11-lp1 login: 5
2021-01-13 07:57:14,273:op-test.common.OpTestHMC:write:DEBUG:
2021-01-13 07:57:14,273:op-test.common.OpTestHMC:write:DEBUG:5
2021-01-13 07:57:14,273:op-test.common.OpTestHMC:write:DEBUG:Password:
2021-01-13 07:57:14,275:op-test.common.OpTestSystem:goto_state:DEBUG:OpTestSystem TRANSITIONED TO: 6
2021-01-13 07:57:14,276:op-test.common.OpTestSOL:run:DEBUG:Starting console
2021-01-13 07:57:14,276:op-test.common.OpTestSSH:connect:DEBUG:sshpass -p passw0rd  ssh -p 22 -l root 9.ab.cd.efg -o PubkeyAuthentication=no -o afstokenpassing=no -q -o 'UserKnownHostsFile=/dev/null'  -o 'StrictHostKeyChecking=no'
2021-01-13 07:57:14,276:op-test.common.OpTestSOL:nontimeout_run:DEBUG:Starting console for new SOL thread
2021-01-13 07:57:16,284:op-test.common.OpTestSSH:write:DEBUG:command-line line 0: Unsupported option "afstokenpassing"
2021-01-13 07:57:16,285:op-test.common.OpTestSSH:write:DEBUG:
2021-01-13 07:57:16,386:op-test.common.OpTestSSH:write:DEBUG:
2021-01-13 07:57:16,386:op-test.common.OpTestSSH:write:DEBUG:~.
2021-01-13 07:57:16,386:op-test.common.OpTestSSH:close:DEBUG:CLOSE Expect Buffer ID=0x7fffa535cf28
2021-01-13 07:57:16,487:op-test.testcases.InstallUpstreamKernel:runTest:DEBUG:Downloading linux kernel
2021-01-13 07:57:16,487:op-test.common.OpTestSSH:connect:DEBUG:sshpass -p passw0rd  ssh -p 22 -l root 9.ab.cd.efg -o PubkeyAuthentication=no -o afstokenpassing=no -q -o 'UserKnownHostsFile=/dev/null'  -o 'StrictHostKeyChecking=no'
2021-01-13 07:57:18,495:op-test.common.OpTestSSH:write:DEBUG:command-line line 0: Unsupported option "afstokenpassing"
2021-01-13 07:57:18,495:op-test.common.OpTestSSH:write:DEBUG:
2021-01-13 07:57:18,596:op-test.common.OpTestSSH:write:DEBUG:
2021-01-13 07:57:18,597:op-test.common.OpTestSSH:write:DEBUG:~.
2021-01-13 07:57:18,597:op-test.common.OpTestSSH:close:DEBUG:CLOSE Expect Buffer ID=0x7fffa48c0f28
2021-01-13 07:57:20,703:op-test:<module>:INFO:Exit with Result errors="1" and failures="0"
2021-01-13 07:57:20,703:op-test.common.OpTestUtil:cleanup:INFO:OpTestSystem Starting to Gather ESEL's
2021-01-13 07:57:20,703:op-test.common.OpTestUtil:cleanup:DEBUG:OpTestSystem encountered a problem trying to gather ESEL's, Exception='NoneType' object has no attribute 'ipmi_sel_elist'
2021-01-13 07:57:20,704:op-test.common.OpTestUtil:dump_versions:INFO:Log Location: /home/jenkins/jenkins_home/workspace/ioci@p10/test-reports/test-run-20210113072941/*debug*
2021-01-13 07:57:20,704:op-test.common.OpTestUtil:dump_versions:INFO:

2021-01-13 07:57:20,705:op-test.common.OpTestUtil:cleanup:DEBUG:self.conf.args.qemu_scratch_disk=None
2021-01-13 07:57:20,705:op-test:<module>:DEBUG:Exit Results with No Failures
2021-01-13 07:57:20,705:op-test:<module>:DEBUG:Exit Results Error #1=
(<testcases.InstallUpstreamKernel.InstallUpstreamKernel testMethod=runTest>, 'Traceback (most recent call last):\n 
 File "/home/jenkins/jenkins_home/workspace/ioci@p10/testcases/InstallUpstreamKernel.py", line 92, in runTest\n    (self.home, self.home))\n  
 File "/home/jenkins/jenkins_home/workspace/ioci@p10/common/OpTestSSH.py", line 225, in run_command\n    return self.util.run_command(self, command, timeout, retry)\n  
 File "/home/jenkins/jenkins_home/workspace/ioci@p10/common/OpTestUtil.py", line 1611, in run_command\n    output = self.try_command(term_obj, command, timeout)\n  
 File "/home/jenkins/jenkins_home/workspace/ioci@p10/common/OpTestUtil.py", line 1631, in try_command\n    pty = term_obj.get_console()\n  
 File "/home/jenkins/jenkins_home/workspace/ioci@p10/common/OpTestSSH.py", line 205, in get_console\n    self.connect(logger)\n  
 File "/home/jenkins/jenkins_home/workspace/ioci@p10/common/OpTestSSH.py", line 185, in connect\n    self.check_set_term()\n  
 File "/home/jenkins/jenkins_home/workspace/ioci@p10/common/OpTestSSH.py", line 201, in check_set_term\n    self, setup_term_flag)\n  
 File "/home/jenkins/jenkins_home/workspace/ioci@p10/common/OpTestUtil.py", line 1477, in setup_term\n    msg="Getting login and sudo not successful, probably connection or credential issue, retry")\ncommon.Exceptions.ConsoleSettings: Setting the prompt or logging in for the console was not successful, check credentials and review the following for more details\nExpect Before Buffer="\r\n~."\nExpect After Buffer="<class \'pexpect.exceptions.EOF\'>" \nMessage="Getting login and sudo not successful, probably connection or credential issue, retry"\n')
2021-01-13 07:58:14,028:op-test.common.OpTestHMC:write:DEBUG:login: timed out after 60 seconds
2021-01-13 07:58:14,279:op-test.common.OpTestSOL:nontimeout_run:DEBUG:Terminating SOL monitoring thread
2021-01-13 07:58:14,279:op-test.common.OpTestSOL:run:DEBUG:Exiting console
2021-01-13 07:58:14,279:op-test.common.OpTestUtil:cleanup:DEBUG:self.conf.args.qemu_scratch_disk=None

```

vijaykumarpuliyala, 2021-02-03:
@harihare I'm seeing same issue on my lpar also . 

Running tests...
----------------------------------------------------------------------


[7mE749: empty buffer[27m

:  runTest (testcases.InstallUpstreamKernel.InstallUpstreamKernel) ... ERROR (20.245s)

======================================================================
ERROR [20.245s]: runTest (testcases.InstallUpstreamKernel.InstallUpstreamKernel)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/var/lib/jenkins/workspace/op-test@2/testcases/InstallUpstreamKernel.py", line 65, in setUp
    self.console_thread = OpSOLMonitorThread(1, "console")
  File "/var/lib/jenkins/workspace/op-test@2/common/OpTestSOL.py", line 63, in __init__
    self.c = self.system.console.get_console(logger=self.logger)
  File "/var/lib/jenkins/workspace/op-test@2/common/OpTestHMC.py", line 487, in get_console
    None, self.system.block_setup_term)
  File "/var/lib/jenkins/workspace/op-test@2/common/OpTestUtil.py", line 1506, in setup_term
    msg="Getting login and sudo not successful, probably connection issue, retry")
common.Exceptions.ConsoleSettings: Setting the prompt or logging in for the console was not successful, check credentials and review the following for more details
Expect Before Buffer=" 
  ^H
^[[7mE749: empty buffer^[[27m^M
:^H
^[[7mE749: empty buffer^[[27m^M
:^H
^[[7mE749: empty buffer^[[27m^M
:PS1=\[console-expect\]#^M
^[[7mE492: Not an editor command: PS1=\[console-expect\]#^[[27m^M
:^H
^[[7mE749: empty buffer^[[27m^M
:^H
^[[7mE749: empty buffer^[[27m^M
:^H
^[[7mE749: empty buffer^[[27m^M
:^H
^[[7mE749: empty buffer^[[27m^M
:"
Expect After Buffer="<class 'pexpect.exceptions.TIMEOUT'>" 
Message="Getting login and sudo not successful, probably connection issue, retry"

2021-02-03 00:43:02,380:op-test:<module>:INFO:Exit with Result errors="1" and failures="0"
2021-02-03 00:43:02,380:op-test.common.OpTestUtil:cleanup:INFO:OpTestSystem Starting to Gather ESEL's
2021-02-03 00:43:02,381:op-test.common.OpTestUtil:dump_versions:INFO:Log Location: /var/lib/jenkins/workspace/op-test@2/test-reports/test-run-20210203004101/*debug*
2021-02-03 00:43:02,381:op-test.common.OpTestUtil:dump_versions:INFO:
----------------------------------------------------------

harihare, 2021-02-04:
I tried on two lpars, on the both the lpars I was able to install upstream kernel and boot, on one lp kernel panic occured due to which test failed. However it was a bug. From framework point of view it was handled appropriately.

@abdhaleegit request to give a try on lp1 or lp2 via jenkins. (make sure ip is changed in the config file)

@vijaykumarpuliyala : may I know the system details where it is failing so that, I can try it out on your system it self.

vijaykumarpuliyala, 2021-02-04:
@harihare I shared machine  details on your slack. Please let me know if you need any other information.

harihare, 2021-02-19:
As per discussion with Abdul and Vijay. The issue is not reproducible. Even for me it work fine on Power 8 Box. 
1. Make sure the system is not in panic state before starting the test

Closing the issue. if you have any concerns or issues. Please feel free to reopen the issue with system details for debugging.

closed 2021-02-19 by harihare