open-power/op-test #597: OPexpect.spawn encountered a problem: Check the lpar activate command

abdhaleegit, 2020-07-14:
```
[console-expect]#2020-07-13 18:01:33,686:op-test.common.OpTestHMC:connect:INFO:Opening the LPAR console
Traceback (most recent call last):
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestHMC.py", line 445, in connect
    raise OpTestError("Check the lpar activate command")
common.OpTestError.OpTestError: Check the lpar activate command

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jenkins/jenkins_home/workspace/ioci/OpTestConfiguration.py", line 837, in objs
    hmc.set_system(self.op_system)
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestHMC.py", line 279, in set_system
    self.console.set_system(system)
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestHMC.py", line 352, in set_system
    self.pty = self.get_console()
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestHMC.py", line 465, in get_console
    self.connect(logger=logger)
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestHMC.py", line 449, in connect
    'OPexpect.spawn encountered a problem: ' + str(exp), -1)
common.Exceptions.CommandFailed: Command 'OPexpect.spawn' exited with '-1'.
Output
OPexpect.spawn encountered a problem: Check the lpar activate command
2020-07-13 18:02:39,948:op-test.common.OpTestUtil:cleanup:INFO:OpTestSystem Starting to Gather ESEL's
2020-07-13 18:02:39,949:op-test.common.OpTestUtil:dump_versions:INFO:Log Location: /home/jenkins/jenkins_home/workspace/ioci/test-reports/test-run-20200713180115/*debug*
2020-07-13 18:02:39,949:op-test.common.OpTestUtil:dump_versions:INFO:
----------------------------------------------------------
OpTestSystem Firmware Versions Tested
(if flashed things like skiboot.lid, may not be accurate)
----------------------------------------------------------
Firmware Versions Unavailable
----------------------------------------------------------
----------------------------------------------------------

Traceback (most recent call last):
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestHMC.py", line 445, in connect
    raise OpTestError("Check the lpar activate command")
common.OpTestError.OpTestError: Check the lpar activate command

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./op-test", line 183, in <module>
    OpTestConfiguration.conf.objs()
  File "/home/jenkins/jenkins_home/workspace/ioci/OpTestConfiguration.py", line 923, in objs
    raise e
  File "/home/jenkins/jenkins_home/workspace/ioci/OpTestConfiguration.py", line 837, in objs
    hmc.set_system(self.op_system)
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestHMC.py", line 279, in set_system
    self.console.set_system(system)
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestHMC.py", line 352, in set_system
    self.pty = self.get_console()
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestHMC.py", line 465, in get_console
    self.connect(logger=logger)
  File "/home/jenkins/jenkins_home/workspace/ioci/common/OpTestHMC.py", line 449, in connect
    'OPexpect.spawn encountered a problem: ' + str(exp), -1)
common.Exceptions.CommandFailed: Command 'OPexpect.spawn' exited with '-1'.
Output
OPexpect.spawn encountered a problem: Check the lpar activate command
[Pipeline] }

klauskiwi, 2020-07-27:
We don't have a PowerVM system readily accessible to check this out.. Perhaps @harihare could take a look or @harish-24 could explain what he meant by 'lpar activate command' ?

harish-24, 2020-07-28:
The message is sightly misleading, It should be "Check lpar console activate command". Error seems to occur when trying to take a console of lpar. i.e.

```
# mkvterm -m <managed-sys> -p <lpar>

 Open in progress  
 
 Open Completed. 
```
`Open Completed` is expected to work always when the managed system is active. If this does not happen, then it means there is an issue with the command when connecting.

@abdhaleegit If you thing the expect string failed due to a timing issue, then that needs to be fixed.

harihare, 2021-02-04:
@abdhaleegit any updates here. I this fixed or still issue exist?

harihare, 2021-02-04:
@abdhaleegit please verify on the latest upstream

harihare, 2021-02-19:
discussed with @abdhaleegit this is related to the console issue, which is not reproduced on the latest code. Hence closing the issue. Feel free to reopen the issue if the issue mentioned persist.

closed 2021-02-19 by harihare