open-power/op-test #606: FastrebootHostStressTorture interrupted by unsupported option and caused the result fail.

Peiyu-Jhong, 2020-10-16:
When run OpTestFastReboot.FastRebootHostStressTorture test for a while, it occurred many "Unsupported option "afstokenpassing" " outputs that the next reboot process will show istep IPL and caused the result to fail.
But in OS RHEL7.6 version, it can run Pass without "Unsupported option "afstokenpassing" " outputs.

SUT OS: RHEL8.2 4.18.0-193.19.1.el8_2.ppc64le
LCB OS: Ubuntu 16.04.6 4.14.0-115.16.1.el7a.ppc64le

command: ./op-test -c mihawk_sut2.conf --run testcases.OpTestFastReboot.FastRebootHostStressTorture

The test log: 
The time occurs the Unsupported option is random and after many 'command-line line 0: Unsupported option "afstokenpassing" ' the SUT will show the istep IPL.

 

> [console-expect]#grep 'RESET: Initiating fast' /sys/firmware/opal/msglog
> grep 'RESET: Initiating fast' /sys/firmware/opal/msglog
> [  432.270461725,5] RESET: Initiating fast reboot 2...^M
> [  658.319617971,5] RESET: Initiating fast reboot 3...^M
> [  887.180444814,5] RESET: Initiating fast reboot 4...^M
> [ 1113.285108203,5] RESET: Initiating fast reboot 5...^M
> [ 1339.436277690,5] RESET: Initiating fast reboot 6...^M
> [console-expect]#echo $?
> echo $?
> 0
> [console-expect]#reboot -ff
> reboot -ff
> Rebooting.
> [  162.459880] kvm: exiting hardware virtualization
> [  165.398688] xhci_hcd 0002:01:00.0: Host halt failed, -110
> [  177.473404] reboot: Restarting system
> ^M~.^M~.command-line line 0: Unsupported option "afstokenpassing"^M
> command-line line 0: Unsupported option "afstokenpassing"^M
> ^M~.command-line line 0: Unsupported option "afstokenpassing"^M
> ^M~.command-line line 0: Unsupported option "afstokenpassing"^M
> ^M~.command-line line 0: Unsupported option "afstokenpassing"^M
> ^M~.command-line line 0: Unsupported option "afstokenpassing"^M
> ^M~.command-line line 0: Unsupported option "afstokenpassing"^M
> ^M~.command-line line 0: Unsupported option "afstokenpassing"^M
> ^M~.command-line line 0: Unsupported option "afstokenpassing"^M
> ^M~.command-line line 0: Unsupported option "afstokenpassing"^M
> ^M~.^L
> command-line line 0: Unsupported option "afstokenpassing"^M
> ^M~.command-line line 0: Unsupported option "afstokenpassing"^M
> ommitId[0x53c17269] ==--
> ^Mistep 3.19
> ^Mistep 3.20

The test result shows the error message: False is not true: Did not do fast reboot

> ======================================================================
> FAIL [2100.984s]: runTest (testcases.OpTestFastReboot.FastRebootHostStressTorture)
> ----------------------------------------------------------------------
> Traceback (most recent call last):
>   File "/home/ooo/0814_op-test/testcases/OpTestFastReboot.py", line 166, in runTest
>     "Did not do fast reboot")
> AssertionError: False is not true: Did not do fast reboot
> 
> ----------------------------------------------------------------------
> Ran 1 test in 2100.985s
> 
> FAILED (failures=1)

klauskiwi, 2020-10-16:
The 'afstokenpassing' shouldn't be failing any tests - instead, looks like you may be hitting this: https://github.com/open-power/skiboot/issues/259

However, you'd probably be hitting this issue regardless of OS.. Can you confirm that the exact same firmware passes this test on RHEL 7.6 but fails on RHEL 8.2?

Thanks

Peiyu-Jhong, 2020-10-19:
Both the RHEL 7.6 and  RHEL 8.2 of the firmware version are the same. Below is the detailed firmware version 
BMC : op940.00.mih-5-0-g86f9791c2
PNOR : IBM-mihawk-ibm-OP9-v2.4-4.37-prod

And the skiboot version is v6.5.2-p06015be we used. 
Thanks.

Gene-Lo, 2020-10-22:
Tag ssant@in.ibm.com
Sachin, please kindly track issue.

klauskiwi, 2020-11-30:
The fastreboot issue was fixed upstream and I re-added the Fastreboot testcase to the default routine: https://github.com/open-power/op-test/pull/609

Please check if you still have issues and update this issue accordingly.

harihare, 2021-02-04:
As per previous comment, I am closing the issue. Please feel free to reopen the same if issue still exist.

closed 2021-02-04 by harihare