open-power/op-test #611: op-test.common.OpTestUtil:try_sendcontrol:WARNING:OpTestSystem recovered from temporary issue, but the command output is unavailable, raised Exception CommandFailed

vijaykumarpuliyala, 2021-02-03:
When I'm trying to install latest kernel on my lpar using op-test, I can see below error messages. 
sh -x ./arch/powerpc/boot/install.sh "5.11.0-rc6+" vmlinux System.map "/boot"
+ set -e
+ '[' -x /root/bin/installkernel ']'
+ '[' -x /sbin/installkernel ']'
+ exec /sbin/installkernel 5.11.0-rc6+ vmlinux System.map /boot
2021-02-02 21:02:46,258:op-test.common.OpTestUtil:try_sendcontrol:WARNING:OpTestSystem detected something, working on recovery
^C^C^M~.
2021-02-02 21:03:03,007:op-test.common.OpTestUtil:try_recover:WARNING:OpTestSystem detected something, working on recovery
command-line line 0: Unsupported option "afstokenpassing"^M
Activate the web console with: systemctl enable --now cockpit.socket

This system is not registered to Red Hat Insights. See https://cloud.redhat.com/
To register this system, run: insights-client --register

Last login: Tue Feb  2 11:01:28 2021 from 9.40.200.43^M
[root@ci-upstream1 ~]# which bash && exec bash --norc --noprofile
PS1=\[console-expect\]#
which bash && exec bash --norc --noprofile
/usr/bin/bash
bash-4.4# PS1=\[console-expect\]#
[console-expect]#which stty && stty cols 300;which stty && stty rows 30
which stty && stty cols 300;which stty && stty rows 30
/usr/bin/stty
/usr/bin/stty
[console-expect]#export LANG=C
export LANG=C
[console-expect]#

[console-expect]#
date
which whoami && whoami

[console-expect]#date
Tue Feb  2 21:03:06 EST 2021
[console-expect]#which whoami && whoami
/usr/bin/whoami
root
[console-expect]#echo $?
echo $?
0
[console-expect]#^C^C
2021-02-02 21:03:10,480:op-test.common.OpTestUtil:try_recover:WARNING:OpTestSystem recovered from temporary issue, continuing
2021-02-02 21:03:10,480:op-test.common.OpTestUtil:try_sendcontrol:WARNING:OpTestSystem recovered from temporary issue, but the command output is unavailable, raised Exception CommandFailed but continuing
[console-expect]#ERROR (36117.043s)

======================================================================
ERROR [36117.043s]: runTest (testcases.InstallUpstreamKernel.InstallUpstreamKernel)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/var/lib/jenkins/workspace/op-test/testcases/InstallUpstreamKernel.py", line 117, in runTest
    onlinecpus, timeout=self.host_cmd_timeout)
  File "/var/lib/jenkins/workspace/op-test/common/OpTestSSH.py", line 225, in run_command
    return self.util.run_command(self, command, timeout, retry)
  File "/var/lib/jenkins/workspace/op-test/common/OpTestUtil.py", line 1616, in run_command
    raise cf
  File "/var/lib/jenkins/workspace/op-test/common/OpTestUtil.py", line 1611, in run_command
    output = self.try_command(term_obj, command, timeout)
  File "/var/lib/jenkins/workspace/op-test/common/OpTestUtil.py", line 1679, in try_command
    output_list, echo_rc = self.try_sendcontrol(term_obj, command)
  File "/var/lib/jenkins/workspace/op-test/common/OpTestUtil.py", line 1148, in try_sendcontrol
    " but the command output is unavailable", -1)
common.Exceptions.CommandFailed: Command 'make -j 41 -s && make modules && make modules_install && make install' exited with '-1'.
Output
run_command TIMEOUT in try_sendcontrol, we recovered the prompt, but the command output is unavailable

2021-02-02 21:03:12,486:op-test:<module>:INFO:Exit with Result errors="1" and failures="0"
2021-02-02 21:03:12,486:op-test.common.OpTestUtil:cleanup:INFO:OpTestSystem Starting to Gather ESEL's
2021-02-02 21:03:12,487:op-test.common.OpTestUtil:dump_versions:INFO:Log Location: /var/lib/jenkins/workspace/op-test/test-reports/test-run-20210202110017/*debug*
2021-02-02 21:03:12,487:op-test.common.OpTestUtil:dump_versions:INFO:
----------------------------------------------------------
OpTestSystem Firmware Versions Tested
(if flashed things like skiboot.lid, may not be accurate)
----------------------------------------------------------
Firmware Versions Unavailable

harihare, 2021-02-04:
@vijaykumarpuliyala  Are you using upstream code and code with localfix?

harihare, 2021-02-04:
is is VM or NV?

vijaykumarpuliyala, 2021-02-04:
@harihare It is Power9 ZZ lpar.

vijaykumarpuliyala, 2021-02-04:
> 
> 
> @vijaykumarpuliyala Are you using upstream code and code with localfix?

I'm running op-test job from my Jenkins.

harihare, 2021-02-19:
As per the issue. it is the issue with the kernel config file or some missing packages that is why the Make is failed. Nothing to do with the framework. Hence closing the issue. 

Please feel free to reopen the issue if you feel issue with the framework.

closed 2021-02-19 by harihare