open-power/op-test #769: Support for running commands on EBMC

Anjali-Karamil, 2023-08-02:
Support for running commands on the BMC of an LPAR is required to automate out-of-band experiments which require running some commands on the EBMC console of the LPAR. For example, I would like to automate an Energy management related experiment using op-test where I need to run the following commands on the EBMC of the LPAR:
1. To set system frequency - `busctl call org.open_power.OCC.Control /org/open_power/control/occ0 org.open_power.OCC.PassThrough SetMode yq 11 <frequency_MHz>`
For example, to set the frequency to 3250 MHz, the command would be `busctl call org.open_power.OCC.Control /org/open_power/control/occ0 org.open_power.OCC.PassThrough SetMode yq 11 3250`
2. To get the power consumed by a component - `busctl get-property org.open_power.OCC.Control /xyz/openbmc_project/sensors/power/<sensor_name> xyz.openbmc_project.Sensor.Value Value`
For example, to get the power consumed by the IO link A, the command would be `busctl get-property org.open_power.OCC.Control /xyz/openbmc_project/sensors/power/io_a_power xyz.openbmc_project.Sensor.Value Value`

PraveenPenguin, 2024-07-10:
@Anjali-Karamil can we close this issue, hope you got unblocked now

Anjali-Karamil, 2024-07-12:
Yes, the issue can be closed now. I am able to use the feature to make progress. 
Thank you for working on this.

closed 2024-07-15 by PraveenPenguin