open-power/op-test #776: Jobs are getting failed in between

bskjois, 2023-08-22:
Whenever I trigger some jobs, it starts smoothly but in between it gets failed with the below error

```
2023-08-21 11:10:50,066:op-test.common.OpTestUtil:try_sendcontrol:WARNING:OpTestSystem detected something, working on recovery
^C^C (037/230) type_specific.io-github-autotest-libvirt.virsh.memtune.negative_test.invalid_3:  INTERRUPTED: Test interrupted: Interrupted by user (349.67 s)
```

Though there is no manual intervention, it fails saying Interrupted by user.

After sometime it says it is recovered. But job execution will not resume.

recent job link: https://ltc-jenkins.aus.stglabs.ibm.com:8080/view/KVM%20CI/job/kvmci/108/console

Below is the log:
----------------------

2023-08-21 11:10:51,662:op-test.common.OpTestUtil:try_sendcontrol:WARNING:OpTestSystem recovered from a temporary issue, continuing
Warning: Permanently added '9.3.101.150' (ECDSA) to the list of known hosts.
scp: /home/kvmci/latest: No such file or directory
2023-08-21 11:10:52,148:op-test.common.OpTestHost:copy_files_from_host:ERROR:Copying /home/kvmci/latest file(s) from host failed
ERROR

======================================================================
ERROR: runTest (testcases.RunHostTest.RunHostTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins/jenkins_home/workspace/kvmci/testcases/RunHostTest.py", line 74, in runTest
    con.run_command(line, timeout=self.host_cmd_timeout)
  File "/home/jenkins/jenkins_home/workspace/kvmci/common/OpTestSSH.py", line 225, in run_command
    return self.util.run_command(self, command, timeout, retry)
  File "/home/jenkins/jenkins_home/workspace/kvmci/common/OpTestUtil.py", line 1753, in run_command
    raise cf
  File "/home/jenkins/jenkins_home/workspace/kvmci/common/OpTestUtil.py", line 1748, in run_command
    output = self.try_command(term_obj, command, timeout)
  File "/home/jenkins/jenkins_home/workspace/kvmci/common/OpTestUtil.py", line 1823, in try_command
    raise CommandFailed(command, res, echo_rc)
common.Exceptions.CommandFailed: Command 'python ./avocado-setup.py --run-suite guest_sanity --guest-os JeOS.27.ppc64le --only-filter 'virtio_scsi virtio_net qcow2' --no-download' exited with '1'.

sacsant, 2023-08-22:
runTest con.run_command(line, timeout=self.host_cmd_timeout) 
The run probably timed out. What is the value specified in the conf file for host_cmd_timeout=

bskjois, 2023-09-20:
Tests are getting executed smoothly now. No issues observed in past few runs. Hence closing this issue.

closed 2023-09-20 by bskjois