open-power/op-test #799: kdump/Fadump testcases shows console issue with few testcases

SACHIN-BAPPALIGE, 2023-12-06:
Test will fail after hitting this message in op-tests with  kdump testsuite  execution

python3 op-test --run-suite osdump-suite -c lpar.conf

"try_recover:WARNING:OpTestSystem recovered from temporary issue, continuing"

====================logs========================================

ltcrain108-lp11 login: 2023-12-05 03:01:57,183:op-test.common.OpTestSSH:get_console:INFO:# Reconnecting
[console-expect]#^M~.Connection to 9.40.193.236 closed by remote host.^M

~.command-line line 0: Unsupported option "afstokenpassing"^M
Last login: Tue Dec  5 14:30:32 2023 from 9.53.174.133^M
^[]0;root@ltcrain108-lp11:~^G[root@ltcrain108-lp11 ~]# which bash && exec bash --norc --noprofile
PS1=\[console-expect\]#
which bash && exec bash --norc --noprofile
/usr/bin/bash
bash-4.4# PS1=\[console-expect\]#
[console-expect]#which stty && stty cols 300;which stty && stty rows 30
which stty && stty cols 300;which stty && stty rows 30
/usr/bin/stty
/usr/bin/stty
[console-expect]#export LANG=C
export LANG=C
[console-expect]#

[console-expect]#
date
which whoami && whoami

[console-expect]#date
Tue Dec  5 14:32:01 IST 2023
[console-expect]#which whoami && whoami
/usr/bin/whoami
root
[console-expect]#echo $?
echo $?
0
[console-expect]#cat /proc/cmdline
cat /proc/cmdline
BOOT_IMAGE=/vmlinuz-4.18.0-526.el8.ppc64le root=/dev/mapper/rhel_ltcrain108--lp11-root ro rd.lvm.lv=rhel_ltcrain108-lp11/root rd.lvm.lv=rhel_ltcrain108-lp11/swap crashkernel=2G-16G:512M,16G-64G:1G,64G-128G:2G,128G-:4G default_hugepagesz=1GB hugepagesz=1GB hugepages=80
[console-expect]#echo $?
echo $?
0
uname -a
uname -a
Password: 2023-12-05 03:02:19,513:op-test.common.OpTestUtil:try_sendcontrol:WARNING:OpTestSystem detected something, working on recovery
2023-12-05 03:02:30,624:op-test.common.OpTestUtil:try_recover:WARNING:OpTestSystem detected something, working on recovery
2023-12-05 03:02:30,624:op-test.common.OpTestHMC:connect:INFO:De-activating the console
rmvterm -m ltcrain108 -p ltcrain108-lp11-CR
rmvterm -m ltcrain108 -p ltcrain108-lp11-CR
Close command sent[console-expect]#echo $?
echo $?
0
[console-expect]#2023-12-05 03:02:30,869:op-test.common.OpTestHMC:connect:INFO:Opening the LPAR console
2023-12-05 03:02:47,087:op-test.common.OpTestUtil:try_recover:WARNING:OpTestSystem recovered from temporary issue, continuing
2023-12-05 03:02:47,087:op-test.common.OpTestUtil:try_sendcontrol:WARNING:OpTestSystem recovered from temporary issue, but the command output is unavailable, raised Exception CommandFailed but continuing
ERROR

======================================================================
ERROR: runTest (testcases.PowerNVDump.KernelCrash_hugepage_checks)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/root/op-test/testcases/PowerNVDump.py", line 1073, in runTest
    boot_type = self.kernel_crash()
  File "/root/op-test/testcases/PowerNVDump.py", line 360, in kernel_crash
    self.c.run_command("uname -a")
  File "/root/op-test/common/OpTestHMC.py", line 1252, in run_command
    return self.util.run_command(self, i_cmd, timeout)
  File "/root/op-test/common/OpTestUtil.py", line 1820, in run_command
    raise cf
  File "/root/op-test/common/OpTestUtil.py", line 1815, in run_command
    output = self.try_command(term_obj, command, timeout)
  File "/root/op-test/common/OpTestUtil.py", line 1883, in try_command
    output_list, echo_rc = self.try_sendcontrol(term_obj, command)
  File "/root/op-test/common/OpTestUtil.py", line 1351, in try_sendcontrol
    raise CommandFailed(command, "run_command TIMEOUT in try_sendcontrol, we recovered the prompt,"
common.Exceptions.CommandFailed: Command 'uname -a' exited with '-1'.
Output
run_command TIMEOUT in try_sendcontrol, we recovered the prompt, but the command output is unavailable

----------------------------------------------------------------------
Ran 2 tests in 1051.986s

FAILED (errors=1)
2023-12-05 03:02:49,091:op-test:<module>:INFO:Exit with Result errors="1" and failures="0"
2023-12-05 03:02:49,149:op-test.common.OpTestUtil:cleanup:INFO:OpTestSystem Starting to Gather ESEL's
2023-12-05 03:02:49,150:op-test.common.OpTestUtil:dump_versions:INFO:Log Location: /root/op-test/test-reports/test-run-20231205024355/*debug*
2023-12-05 03:02:49,150:op-test.common.OpTestUtil:dump_versions:INFO:
----------------------------------------------------------
OpTestSystem Firmware Versions Tested
(if flashed things like skiboot.lid, may not be accurate)
----------------------------------------------------------
Firmware Versions Unavailable
----------------------------------------------------------
----------------------------------------------------------

[root@ltcden9-lp4 op-test]#

PraveenPenguin, 2024-07-10:
@SACHIN-BAPPALIGE are you still looking help here ?

SACHIN-BAPPALIGE, 2024-08-02:
@PraveenPenguin This issue got resolved with cleanup patch merged to kdump/fadump.  Please close this issue

closed 2024-08-05 by PraveenPenguin